ros-visualization/rviz #1184: PointCloud XYZRGBL RGB8 colour transformer does not show cloud

heuristicus, 2018-01-09:
I have a pointcloud with XYZRGBL fields generated in PCL, whose display does not seem to work correctly in rviz. If I view the cloud in pcl_viewer I can see the following colour handlers:

```
List of available color handlers for actor out.pcd-0: [random](1) x(2) y(3) z(4) rgb(5) label(6)
```
The cloud (attached below) has specific RGB values for each label. However, when displaying this cloud in rviz, if the RGB8 colour transformer is selected, nothing appears. Selecting any of the other colour transformers shows that there are indeed points coming through. With the intensity transformer, using the `label` channel correctly displays the cloud, with different label regions having different colours.

This is in rviz 1.12.13, on Ubuntu 16.04. I'd like to help fix this but there are no error or warning messages which I can use to point me to where the problem might lie.

[out.pcd.zip](https://github.com/ros-visualization/rviz/files/1615482/out.pcd.zip)

jediofgever, 2019-12-30:
having same issue after 2 years

rhaschke, 2019-12-30:
@jediofgever, you are cordially invited to debug this issue and provide a fixing PR.