ros-visualization/rviz #1185: rviz crashes due to invalid quaternion after Interactive Marker Server applyChanges to existing marker

cassinaj, 2018-01-16:
Changes made to the interactive marker via InteractiveMarkerServer::insert() or InteractiveMarkerServer::clear() followed by InteractiveMarkerServer::applyChanges() cause rviz to crash. When changes are made to a marker, an invalid quaternion is passed to OGRE causing the crash. This doesn't seem to affect creating new markers, though. E.g. the crash happens when removing some or even all controllers of a marker and applying the changes resulting in the following server update on the /interactive_marker/update_full topic (both cases the server is not sending unnormalized quaternions):

Clearing server and apply changes:
```
server_id: "/particle_tracker"
seq_num: 2
markers: []
```
Removing a subset of controllers and apply changes:
```
server_id: /particle_tracker
seq_num: 2 
markers: 
  -
    header: 
      seq: 0
      stamp: 
        secs: 1516072361
        nsecs: 227513561
      frame_id: "camera_depth_optical_frame"
    pose: 
      position: 
        x: 0.0934688001871
        y: 0.127441003919
        z: 0.961609005928
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    name: "0"
    description: "[inactive] Click to active"
    scale: 0.20000000298
    menu_entries: []
    controls: 
      - 
        name: "button_control"
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
        orientation_mode: 0
        interaction_mode: 2
        always_visible: True
        markers: 
          - 
            header: 
              seq: 0
              stamp: 
                secs: 0
                nsecs:         0
              frame_id: ''
            ns: ''
            id: 0
            type: 10
            action: 0
            pose: 
              position: 
                x: 0.0
                y: 0.0
                z: 0.0
              orientation: 
                x: 0.0
                y: 0.0
                z: 0.0
                w: 1.0
            scale: 
              x: 1.0
              y: 1.0
              z: 1.0
            color: 
              r: 0.5
              g: 0.5
              b: 0.5
              a: 0.20000000298
            lifetime: 
              secs: 0
              nsecs:         0
            frame_locked: False
            points: []
            colors: []
            text: ''
            mesh_resource: "package://object_meshes/object_models/impact_battery.obj"
            mesh_use_embedded_materials: False
        independent_marker_orientation: False
        description: ''
```

wjwwood, 2018-01-17:
What version of `rviz` and `interactive_markers` are you using, what version of ROS?

> When changes are made to a marker, an invalid quaternion is passed to OGRE causing the crash.

Do you have a traceback? If so please post it. If not then can you tell us how you know this is the source of the crash?

cassinaj, 2018-01-17:
I'm currently using rviz version 1.12.14 using Kinetic on 16.04 with Ogre 1.9. Tested also the indigo branch and compiled it and I get the same issue. On 14.04, indigo and Ogre 1.8 this is not an issue. So it seems to be related to ogre 1.9 which is more strict with invalid quaternions. It took me a while to figure out where the issue was.
First Issue was the one referenced [#1179](https://github.com/ros-visualization/rviz/pull/1179). The interactive marker was rejected due to invalid quaternion. But looking at the InteractiveMarkerServer messages, the orientation of the markers had a valid quaternion, except the control orientation. They are according to the InteractiveMarker tutorials by definition not used as quaternions to determine the orientation of the control. So it seems the quaternion check broke all interactive marker and also made the tutorials invalid. I pulled the latest changes of kinetic version since the quaternion check was reverted and the interactive marker worked until I a change was applied on the InteractiveMarkerServer marker side. I checked the difference between the initial state of the interactive marker and the state when it was updated (applyChanges) and made sure that they match exactly and the crash would still happen. The crash happens when the interactive marker server is destroyed and rviz attempts to remove the markers. With the latest pull a newly added warning about invalid quaternion appears but as you can see above, no invalid quaternion is sent from the server. The traceback was crashing on Ogre level. I will post a traceback tomorrow when I get to it.

cassinaj, 2018-01-18:
Here is a traceback and the warning

```
[ WARN] [1516247964.309571998]: Interactive marker '0' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00000000400a2c6a in ?? ()
(gdb) bt
#0  0x00000000400a2c6a in ?? ()
#1  0x00007fffbae6b809 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.387.26
#2  0x00007fffbae72374 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.387.26
#3  0x00007fffbaa3b295 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.387.26
#4  0x00007fffb8ae19c5 in Ogre::GLRenderSystem::_render(Ogre::RenderOperation const&) () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007ffff39597f7 in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff3983fb9 in Ogre::QueuedRenderableCollection::acceptVisitorGrouped(Ogre::QueuedRenderableVisitor*) const () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff3984095 in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff39428df in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff3942757 in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff3958474 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff3994f51 in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#12 0x00007ffff393977a in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff39395cb in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff393911e in Ogre::RenderTarget::updateImpl() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff3939ac4 in Ogre::RenderTarget::update(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff39e9e5f in Ogre::RenderSystem::_updateAllRenderTargets(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff3b004ba in Ogre::Root::_updateAllRenderTargets() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff3b005a0 in Ogre::Root::renderOneFrame() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#19 0x00007ffff7b4b34e in rviz::VisualizationManager::onUpdate() () from /home/jissac/workspace/dbot_ws/devel/lib/librviz.so
#20 0x00007ffff1f44d2a in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff1f515c8 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff1f45bb3 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff745705c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff745c516 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff1f1638b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff1f6b5ed in QTimerInfoList::activateTimers() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff1f6baf1 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff146d197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff146d3f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff146d49c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff1f6c7cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff1f13b4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007ffff1f1bbec in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x0000000000400c89 in main ()
```

wjwwood, 2018-01-19:
Thanks for the additional details.

One of us will try to look into it (especially reproducing the idea that the tutorial is invalid) as soon as possible. But please continue to post details you find out if you're still looking into the issue.

artivis, 2018-07-16:
Any update on this matter ??
Any workaround ??

Tianhuanyu, 2018-11-12:
I have a same problem when i drag the marker. It reports "Interactive marker 'EE:goal_ee_link' contains unnormalized quaternions". Is there any solusion?

yt100323, 2019-03-23:
any results? i have also run to the problem

Chunting, 2019-03-26:
Any updates? apparently I'm not the only one who have this issue.

martinmeinke, 2019-05-03:
same here, any further information on those issues?! (Bionic & Melodic)

cangjiaxuan, 2019-07-16:
Same problem here, using ubuntu 16.04 + ROS kinetic with ikfast inverse kinematic plugin

rhaschke, 2019-07-16:
@cassinaj, I tried to reproduce your issue on both Melodic and Kinetic, but I wasn't able to do so.
I just started interactive_marker_tutorials' basic_controls and stopped it again (via Ctrl-C) - no problems.
Would you mind to provide a full python or C++ source file to reproduce the issue?

@Tianhuanyu and most others: You are reporting a different issue (the unnormalized quaternion warning). rviz isn't crashing due to that, is it? Please report this issue in MoveIt as it is probably an invalid quaternion send by the MoveIt plugin.

rhaschke, 2019-08-30:
Closing due to inactivity. We have had a lot of improvements regarding marker validation in the past.

closed 2019-08-30 by rhaschke

jmonticelli, 2022-01-05:
This error message was also, coincidentally, the last thing I saw before a segfault that was related to ogre 1.9, as mentioned in this issue: https://github.com/ros-visualization/rviz/issues/1082

See comment:
https://github.com/ros-visualization/rviz/issues/1082#issuecomment-452586831