ros-visualization/rviz #1188: librviz:Image Displays can not show in one window?

sunyuzhe2017, 2018-01-27:
As I learned the librviz tutorial, I added a rviz/Image in the code like this `manager_->addDisplay("rviz/Image",true)
img_ = manager_->createDisplay("rviz/Image","image", true);
ROS_ASSERT( img_ != NULL );
img_->subProp("Image Topic")->setValue("camera/image");`
however,the Image show in another window,how could I integrate it with the main window?Is there any simple tutorials?I have searched for a long time ,find some similar questions([ros answer question](https://answers.ros.org/question/37887/librviz-imagedisplay/)) at ROS answers,but all that don't have a right answer for me .

wjwwood, 2018-01-28:
I don't know, it sounds like you just need to tell Qt to make sure the window is docked on creation, but I don't know how to do that off hand. I would ask you to ask this question on answers.ros.org, perhaps someone there has done this before or has time to look into the solution.

See our support guidelines: http://wiki.ros.org/Support

If it turns out we need a code change in rviz, I can reopen this until a pr can be made to add such an option.

closed 2018-01-28 by wjwwood

sunyuzhe2017, 2018-01-29:
OK，thanks ！