ros-visualization/rviz #1192: rviz segfault when using interactive markers and line_strip or line_list nvidia only

rlober, 2018-02-01:
I am not sure how related this issue is to #1082 and #1185 so it may be a duplicate, but it seems more specific than those.

## Issue
On a computer with nvidia graphics card/driver when you have any interactive markers in the scene and any regular line type markers (`LINE_LIST`, `LINE_STRIP`) when you click on the interactive marker, `rviz` segfaults (see below for function call trace).

**This only happens with `LINE` type markers and is not a problem on non-nvidia computers.** 

## Setup info
```
ubuntu 16.04
nvidia driver: version 384.111
ros: kinetic
rviz branch: kinetic-devel 
rviz commit: 16933c222112a10a77fe771b55cb7cf1e517928e
compiled against: Qt version 5.10.0 and OGRE version 1.9.0
OpenGl version: 4.5
```

## How to replicate
Modify [line 53](https://github.com/ros-visualization/visualization_tutorials/blob/kinetic-devel/interactive_marker_tutorials/src/simple_marker.cpp#L53) of `interactive_marker_tutorials/src/simple_marker.cpp` to use the `frame_id`, `/my_frame`. I.e.

```
int_marker.header.frame_id = "base_link";
```
to
```
int_marker.header.frame_id = "/my_frame";
```
This will allow the simple interactive marker to be displayed along with the markers from the [`points_and_lines.cpp` demo](https://github.com/ros-visualization/visualization_tutorials/blob/kinetic-devel/visualization_marker_tutorials/src/points_and_lines.cpp)

Run these two nodes and add their visualization topics to the `rviz` scene, don't forget to change the `Fixed Frame` to `my_frame`.

When you click on the interactive marker controls, rviz should segfault and crash. You may have to release the mouse click before the crash occurs.

## gdb segfault call trace
```
Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00000000400a2c6e in ?? ()
(gdb) where
#0  0x00000000400a2c6e in ?? ()
#1  0x00007fffeab9a219 in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.111
#2  0x00007fffeaba0d84 in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.111
#3  0x00007fffea768a5e in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.111
#4  0x00007fffb40ca9c5 in Ogre::GLRenderSystem::_render(Ogre::RenderOperation const&) ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007ffff2dc87f7 in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff2df2fb9 in Ogre::QueuedRenderableCollection::acceptVisitorGrouped(Ogre::QueuedRenderableVisitor*) const () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff2df3095 in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff2db18df in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff2db1757 in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff2dc7474 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff2e03f51 in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) ()
---Type <return> to continue, or q <return> to quit---
  -gnu/libOgreMain.so.1.9.0
#12 0x00007ffff2da877a in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff2da85cb in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff2da811e in Ogre::RenderTarget::updateImpl() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff2da8ac4 in Ogre::RenderTarget::update(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff2e58e5f in Ogre::RenderSystem::_updateAllRenderTargets(bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff2f6f4ba in Ogre::Root::_updateAllRenderTargets() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff2f6f5a0 in Ogre::Root::renderOneFrame() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#19 0x00007ffff7aca394 in rviz::VisualizationManager::onUpdate() () from /home/ryan/ws/rviz_ws/install/lib/librviz.so
#20 0x00007ffff7aec8f9 in rviz::VisualizationManager::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
   from /home/ryan/ws/rviz_ws/install/lib/librviz.so
#21 0x00007ffff6a14bc9 in QMetaObject::activate(QObject*, int, int, void**) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#22 0x00007ffff6a213e7 in QTimer::timeout(QTimer::QPrivateSignal) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#23 0x00007ffff6a216b8 in QTimer::timerEvent(QTimerEvent*) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#24 0x00007ffff6a15adb in QObject::event(QEvent*) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#25 0x00007ffff701254c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Widgets.so.5
#26 0x00007ffff7019897 in QApplication::notify(QObject*, QEvent*) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Widgets.so.5
#27 0x00007ffff69e99f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#28 0x00007ffff6a3d08e in QTimerInfoList::activateTimers() () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#29 0x00007ffff6a3d871 in ?? () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#30 0x00007fffee475197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007fffee4753f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007fffee47549c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff6a3dbaf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#34 0x00007fffdd1c89a1 in ?? () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5XcbQpa.so.5
#35 0x00007ffff69e826a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#36 0x00007ffff69f0da4 in QCoreApplication::exec() () from /home/ryan/Qt/5.10.0/gcc_64/lib/libQt5Core.so.5
#37 0x00000000004012ab in main ()
```

rlober, 2018-02-08:
ping

croesmann, 2018-10-10:
I have the same issue with line list markers in combination with interactive markers.

rhaschke, 2019-09-23:
I can confirm this behavior: AddressSanitizer reports access to a null pointer (in the nvidia libs).
No idea, how this is related to interacting with (clicking on) the marker.

rhaschke, 2019-09-23:
Duplicate of #1082. Fixed with an Ogre 1.9.1 source build. As this is not an rviz, but an Ogre issue, I'm closing the issue here.

closed 2019-09-23 by rhaschke