ros-visualization/rviz #1195: Programmatically setting/changing the configuration.

anders29, 2018-02-12:
I've been considering the idea of adding functionality to RViz which allows programmatic modification of the config. This would likely consist of a service call /rviz/setConfig which takes a filepath to an  RViz config file. One could perhaps further with services to load/close specific plugins.

I wanted to get input before proceeding since it's a pretty big departure from the existing method of interaction with the program.

wjwwood, 2018-02-12:
It's something that's come up a few times in the past. I think that a service to load a config is fine. For saving a config, it would probably be ok, but "save as" gets a bit tricky, because you have to handle cases like permissions, folder existence, security concerns, etc... I suppose, like load, it could just succeed or fail without dealing with any of those issues.

Either way, load and save are pretty straightforward in my opinion.

As for more granulated control of the user interface via ROS, that's also come up, and I think it would be really neat to have, but I would post a relatively in depth proposal for that before spending a lot of time on it.

anders29, 2018-02-12:
Good to hear I'm not alone on this. I'm thinking that for now I'll target a loadConfig and a shallow saveConfig. A deeper saveConfig and more granular controls could come later. I'll propose an API in more detail once I've investigated the relevant code.

anders29, 2018-02-14:
After looking things over, it appears we need to expand the VisualizationFrame class for this functionality. A new load service callback method would call VisualizationFrame::loadDisplayConfig, and the save service callback method would call VisualizationFrame::saveDisplayConfig.

These could be the service definitions:
~/LoadConfig
LoadConfig.srv
     std_msgs/String path # fully resolved path to .rviz file
---
     bool success

~/SaveConfig
SaveConfig.srv
     std_msgs/String path # path to save location
---
     bool success

To signal success or failure, we'll also need to change VisualizationFrame::loadDisplayConfig to return bool instead of void.

Later on, we could implement loading individual Panels by hooking a service callback up to VisualizationFrame::addPanelByName.

wjwwood, 2018-02-15:
Those definitions look good to me (I edited your comment for style). I thought maybe we'd want a string in addition to `bool success` so you could communicate the reason for the failure, but they can just check the console of rviz for that I guess.

> To signal success or failure, we'll also need to change `VisualizationFrame::loadDisplayConfig` to return `bool` instead of `void`.

Does it not raise an exception instead?

> Later on, we could implement loading individual Panels by hooking a service callback up to `VisualizationFrame::addPanelByName`.

Seems like a logic next step, but to be complete you'd also want to be able to list panels and remove them or something.

anders29, 2018-02-15:
> Does it not raise an exception instead?

I'm not seeing any throw statements in the function. If the input path is invalid for some reason, it instead tries to load the default config file. If that fails too then it prints to ROS_ERROR and returns void. This does raise the question of whether we want to change that behavior. If the service sends an invalid path I don't think we want RViz to switch to the default config, but rather just return false.

wjwwood, 2018-02-15:
You might need a different function then, rather than changing this one (though they could call one another or share yet another function).

anders29, 2018-02-15:
Working on this now. I realized we don't actually need two srv definitions since they're identical. We can name it something more generic like SendFilePath.srv and use it in both services.