ros-visualization/rviz #1197: MovableText Linespacing not working

ecazaubon, 2018-02-15:
Hello,
I have noticed a problem with the Line spacing in rviz::MovableText, it is not taken into account. 
Setter function exists but the variable is not used in the logic itself and instead 0.01 is used for top corners calculation only. 

Also, I do not know if it is a bug or not but for the calculation of the AABB, if the alignment is V_CENTER or V_ABOVE, the AABB will be wrong, due to the fact that the 0.01 line spacing is taken into account by the calculation of the top point but not in the drawing/calculation of the bottom [here in the file](https://github.com/ros-visualization/rviz/blob/fe0512e6ce3c7d4ae54d57ec11c8cdd87acca4d5/src/rviz/ogre_helpers/movable_text.cpp#L377)

wjwwood, 2018-02-15:
Sounds like a bug. I don't have time to investigate it, but a pull request would be appreciated if you (or someone else) manage to fix it yourselves.

dhood, 2018-09-17:
closed by https://github.com/ros-visualization/rviz/pull/1200

closed 2018-09-21 by ecazaubon