ros-visualization/rviz #1201: Loading TIFFs in Meshes reports non-fatal errors

DLu, 2018-02-20:
Upstream: [Error in libtiff5](https://bugs.launchpad.net/ubuntu/+source/tiff/+bug/1681552)
Downstream: [pr2_description](https://github.com/PR2/pr2_common/issues/257) [urdf_tutorial](https://github.com/ros/urdf_tutorial/issues/27)

Reported in xenial/kinetic.

wjwwood, 2018-03-20:
Given https://bugs.launchpad.net/ubuntu/+source/tiff/+bug/1681552, is there anything I can do in rviz? I don't think that there is, but did you have an idea or suggestion?

DLu, 2018-03-20:
Removing the dependency is one option, but really, I just wanted to put the bug report in the "right" ROS repo.

wjwwood, 2018-03-20:
> Removing the dependency is one option, but really, I just wanted to put the bug report in the "right" ROS repo.

rviz doesn't use tiff, it's either used in Ogre or assimp or something else, but I don't think I can remove the dependency.

I'll close this because it has to get fixed in the ubuntu issues you linked or it has to be removed in one of our dependencies.

So I'm going to close this, but if anyone has an idea of what I could change in rviz to prevent this let me know and I'll reopen it.

closed 2018-03-20 by wjwwood