ros-visualization/rviz #1202: Expose Rviz Viewpoint Transformation

rkeatin3, 2018-02-23:
It would be a nice feature to be able to get the transformation of the Rviz viewpoint w.r.t. the fixed frame. Teleoperation is a good use case for this, as it can be difficult/unintuitive to control a robot w.r.t. a fixed frame when changing to viewpoint. 

Thoughts?

wjwwood, 2018-02-23:
Sounds reasonable to me, only question is in which ways:

- programmatically (via a plugin)
- a ROS topic
- tf broadcaster

Personally, I'd like to see a new kind of plugin (maybe a "ViewControllerObserver" plugin) which has access to this information, but that would just be "nice to have" (following the dog fooding principal). That plugin interface could then be used to implement a plugin for that interface which publishes to a topic and/or does a tf broadcast. I think the ros topic could be "always on", but for the tf broadcaster you would definitely want to have that opt-in only.

Original ROS answers question: https://answers.ros.org/question/283462/rviz-viewpoint-transform/

caomuqing, 2019-10-01:
Is this feature implemented? Or is there any other way to get the transformation of rviz viewpoint?
Thanks!

rhaschke, 2019-10-01:
No, as far as I know, this feature is not yet implemented. But you are invited to so.

caomuqing, 2019-10-01:
Hi rhaschke,

Thanks for your information. I would love to help implementing this feature, but I have not read any rviz code before. If anyone can provide some hints on where I should start looking into, or maybe a basic outline of steps to do this implementation, that would be very helpful.

Muqing

caomuqing, 2019-10-02:
For those interested I did a quick modification of the rviz_animated_view_controller so that it publishes the rviz view position and orientation as a pose message at 10Hz. I tested it under ROS Kinetic. Not a very neat solution but sufficient for people who just want this piece of information.
You are welcome to try it [here](https://github.com/caomuqing/rviz_view_publisher).

Best Regards,
Muqing