ros-visualization/rviz #1203: rviz displays in only a quarter of the display area

Dalchemisy, 2018-03-03:
I compiled rviz on OS X 10.13.3 with the latest commit, and it can be launched, however rviz displays only in the left bottom quarter of the display area:

[image]

When I resize the display area, the part that showing the image remains always a quarter of it. The message in terminal is listed below"

rosrun rviz rviz
\[ INFO] [1520065250.430320000]: rviz version 1.12.15
[ INFO] [1520065250.430372000]: compiled against Qt version 5.10.0
[ INFO] [1520065250.430381000]: compiled against OGRE version 1.7.4 (Cthugha)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1520065250.733799000]: Stereo is NOT SUPPORTED
[ INFO] [1520065250.733862000]: OpenGl version: 2.1 (GLSL 1.2).
libpng warning: iCCP: known incorrect sRGB profile

Could anyone give me some hint about how to fix this? Thanks a lot in advance!

boris-il-forte, 2018-03-06:
i think this is related to #1069

rhaschke, 2019-09-24:
Fixed via #1263, released in 1.13.4.

closed 2019-09-24 by rhaschke