ros-visualization/rviz #1204: Display links' centers of mass

Garrus007, 2018-03-06:
I add visualization of links' centers of mass in RobotModel display type.
It adds spheres in intertial origin positions.

Here it is:
[image]
(in this example urdf inertial origins is incorrect, don't worry)

Do you need this functionality?

jlack1987, 2019-01-04:
I think additionally it would be nice to show inertial approximations with an ellipsoid

chapulina, 2019-01-04:
As a reference, Gazebo displays the center of mass as a sphere sized as if it were a ball of lead with the same mass as the link, and the inertia as a box sized as if it were a box of uniform density that has the same inertia as the link.

An example:

[image]

[Here](https://bitbucket.org/osrf/gazebo/src/582827025ea59c832cc50e33d64a07ae13cad76a/gazebo/rendering/COMVisual.cc?fileviewer=file-view-default#COMVisual.cc-130)'s the CoM calculation.
[Here](https://bitbucket.org/osrf/gazebo/src/582827025ea59c832cc50e33d64a07ae13cad76a/gazebo/rendering/InertiaVisual.cc?fileviewer=file-view-default#InertiaVisual.cc-61)'s the inertia calculation, which uses [ignition::math::MassMatrix3::EquivalentBox](https://bitbucket.org/ignitionrobotics/ign-math/src/c86dbbcbee2e892af289ef0cf2dc8c0c0b463e73/include/ignition/math/MassMatrix3.hh?fileviewer=file-view-default#MassMatrix3.hh-1003)

jlack1987, 2019-01-04:
Yeah something similar would be great for the `RobotModel`. I like that Gazebo shows, would be great if we had something on similar for when we're on hardware

luator, 2019-12-02:
Is anyone working on this? It would be very useful feature.

Garrus007, 2019-12-02:
I have a [commit](https://github.com/vstuhumanoid/rviz/commit/f215b91db6fd673b8569ef61c7c3a4001fa20cc1) in my old forked repo. But there are several bugs (It can segfault because I forgot to check several parameters). I can update this to the current version of the RViz and create Pull-Request in few days.

rhaschke, 2019-12-02:
You are very welcome to provide your solution and update with it with the pointers provided by @chapulina.

Yeshasvitvs, 2020-03-12:
@Garrus007 do you have any update on this ?

JuliusSustarevas, 2021-02-26:
Is there something stopping this? Or just bandwidth? I'd especially be interested int he full center of mass of the whole robot model

briansoe66, 2021-06-23:
@Garrus007 @Yeshasvitvs @JuliusSustarevas I have created a Pull request for ros2 rviz, based on @Garrus007 fork. I would appreciate help testing the implementation, and understanding the failed checks. Let me know if you can help and want edit write access to the fork/branch.

briansoe66, 2021-07-19:
ros2 rviz can now display links' center of mass and inertia, as of commit 2b551f808e8e8c9bd29bc006d06a549c00db4861

jose-adrian, 2021-10-17:
Hi guys, as per the last reply, ros2 rviz can now display links' center of mass and inertia, is there something special that needs to be done in order for this to work? I'm a beginner testing some simulations from my university and the Center of Mass check box does not appear in my RobotModel.