ros-visualization/rviz #1206: Viewport interaction from Panel objects

jr-bor, 2018-03-08:
I am doing a Panel plugin where I would like to have inputs from the Viewport. To put some context, I would like to receive the mouse events (position, click type and button) so I can "draw" polygons in the map. After digging around I saw (please, correct me if I am wrong) that the one that handles the mouse clicks from the RenderPanel is the VisualizationManager::handleMouseEvent. This one, forwards the event to the active Tool from the ToolManager. Thus, the only way of interacting with the Scene is by inheriting from Tool. The problem is that my Panel is not a Tool :(.

I could not come with many solutions, being the least invasive and flexible the one in PR #1205. It has two major problems though:

1. Modifies Rviz, which is not ideal because I need my colleagues to compile Rviz I want them to use the plugin.
2. Introduces the use of Qt signals/slots, which I see it goes somehow against how Rviz architecture is.

My doubts are:

1. Is there any other way to interact from a Panel with the mouse in the Viewport?
2. Why Qt signals are not emitted from the VisualizationManager so the rest of the Qt objects can listen from them? Which extending the question, why Qt signals/slots mechanism are not used all around Rviz?

wjwwood, 2018-03-08:
> Is there any other way to interact from a Panel with the mouse in the Viewport?

Not that I'm aware of, basically it does this:

QMouseEvent -> Active Tool -(move tool)> View Controller

Some how the Interactive markers gain access to this, but I don't remember how they work off hand, but you could look at them to see if you could piggyback off of that.

> Why Qt signals are not emitted from the VisualizationManager so the rest of the Qt objects can listen from them? Which extending the question, why Qt signals/slots mechanism are not used all around Rviz?

No idea, I'm not the original author, but I guess it just wasn't needed so it wasn't done. One thing I will say is that some care is taken to use signals and slots where gui elements are involved so that it happens in the qt thread, but I don't know why other things would not use signals and slots. One thought is that they wanted some things to be done outside of the qt thread to avoid slowing down interactions, but this particular thing (handling the mouse event) is already in that thread.

So I'd say that there's no good reason behind it.

jr-bor, 2018-03-08:
Thanks for the fast reply!

> Some how the Interactive markers gain access to this, but I don't remember how they work off hand, but you could look at them to see if you could piggyback off of that.

Interactive markers and [other plugins](https://www.youtube.com/watch?v=3W1ufJ7rpCA&feature=youtu.be) as far as I am aware they "get out of rviz" by publishing the info in a topic that is then recovered. Feasible, but a workaround.

Based on your comment in the PR #1205 I will do a Tool specifically to "draw" in the map and I will connect the panel and the tool either with a Qt signal/slot or with a topic.

Thanks for the help!

closed 2018-03-19 by jr-bor