ros-visualization/rviz #1208: Links with Multiple Visual Elements with Different Colors Don't Render Correctly

TimCraig, 2018-03-13:
Using RViz 1.12.15, if a link has multiple visual elements, the first visual element renders with the specified material color.  The remaining  visual elements render as black.  The model I used was processed with Xacro but I examined the generated URDF and the materials were specified correctly for the link.

dhood, 2018-04-07:
@TimCraig could you provide a reproducible example? That will help this be investigated, thanks!

TimCraig, 2018-04-10:
Here's a simple file with two links and one joint.  The base_link has 3 visuals each consisting of a cylinder 1 meter tall and half a meter in diameter spaced along the x-axis.  They should be red, green, and blue.  The green one in the middle displays properly while the other two show up in RViz as black.  I've included a screenshot.   Running ROS Kinetic on Ubuntu Mate 16.04 in a VirtualBox on Windows 10.

[color_problem.zip](https://github.com/ros-visualization/rviz/files/1893101/color_problem.zip)

dhood, 2018-04-17:
thanks for the example, we will test it out when we get a chance

rhaschke, 2019-03-08:
This was fixed in #1079 and released into 1.12.5

closed 2019-03-08 by rhaschke