ros-visualization/rviz #1214: Improper working of Interactive marker in rviz

mukesh148, 2018-04-03:
Hello all,
I am using moveit setup tutorial for kinetic version. Instead of pr2 robot, I have generated config file for my own robot. But when I was visualized it in rviz it shows following error -

`Interactive marker 'EE:goal_link5' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.`

I am attaching my config file also, any help is appreciated. Thanks.
[https://drive.google.com/open?id=1cPfW2JxU0rGC3Bdub8vKyUQd5BIHuSCj](url)

dhood, 2018-04-17:
Hi @mukesh148, this is not an error with your custom setup (or with rviz) but it's a warning caused by one of the plugins publishing an invalid quaternion. If you identify the publisher of the invalid quaternion please notify the maintainers of the package.

I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask that in the future you raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rationale for asking all kinds of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

closed 2018-04-17 by dhood

aydal, 2018-05-29:
Hi @mukesh148, did you resolve the error?