ros-visualization/rviz #1219: rviz crashing with rviz_imu_plugin

davidbsp, 2018-04-09:
I'm running ROS Kinetic on Ubuntu 16.04 and getting a crash whenever I want to visualize an IMU object in rviz with the rviz_imu_plugin.

(Different) errors I get:

```
info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[rviz-4] process has died 
```

```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
```

```
rviz: tpp.c:84: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.
[rviz-3] process has died
```

Rviz Log:
```
[ INFO] [1523291843.839827231]: rviz version 1.12.4
[ INFO] [1523291843.839859169]: compiled against Qt version 5.5.1
[ INFO] [1523291843.839865791]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1523291844.557061127, 1520362440.087018622]: Stereo is NOT SUPPORTED
[ INFO] [1523291844.557128263, 1520362440.087018622]: OpenGl version: 4.6 (GLSL 4.6).
```

dhood, 2018-04-09:
Could you please raise this issue with the maintainers of the rviz_imu_plugin at https://github.com/ccny-ros-pkg/imu_tools/tree/11f405e89925d75b9a17c64d92a139a10ae56118 with a backtrace and reproducible example.
If the cause is identified as an issue within rviz itself we can reopen it here.

closed 2018-04-09 by dhood

davidbsp, 2018-04-10:
Moved it to: [https://github.com/ccny-ros-pkg/imu_tools/issues/86](https://github.com/ccny-ros-pkg/imu_tools/issues/86)
Thanks.