ros-visualization/rviz #1220: TF and other markers all black

HWiese1980, 2018-04-11:
Hi everyone,

I'm experiencing another weird visualization issue. Markers indicating TF reference frames and other markers of CUBE and SPHERE type (haven't checked other mesh markers) are all black, no matter what color I give them. 

I would like to create an ogre.log to be able to provide useful information but for some reason `rviz -l` does nothing. 

Is this a known issue?

dhood, 2018-04-11:
Users have reported seeing meshes as white recently: https://github.com/ros-visualization/rviz/issues/1120

Could you have a read through that thread and see what similarities there are with your situation?

Could you also provide your ROS and RViz versions, and an example mesh + code?

dhood, 2018-04-11:
It's a long shot, but you don't happen to have the "default light" unchecked in the Global Options do you? (that will make everything show black)

HWiese1980, 2018-04-13:
Yepp, "default light" was in fact switched off. No idea why. Either it happend on its own or I clicked that check box without noticing... anyway, my colors are back, everything's fine, no bug.

Thanks for you help! :+1:

closed 2018-04-13 by HWiese1980

HWiese1980, 2018-04-13:
Sure. I don't really like it when people just run off with the answer to never be heard of again. Credit where credit is due. ;-)