ros-visualization/rviz #1222: meshes of RobotLinks are not cached and reused?

rhaschke, 2018-04-14:
When enabling the robot train in MoveIt's MotionPlanning plugin, I observe a dramatic increase of memory usage that is linear with the number of steps in the trail. However, all these visualizations use the same meshes and they should be reused. Obviously they are not. In my case, with a complex robot model, my machine memory of 12GByte is exhausted already with >650 trail steps.

dhood, 2018-04-17:
Visualisations using `loadMeshFromResource` will [reuse existing meshes](https://github.com/ros-visualization/rviz/blob/7970ba08cee3810cfa1609c3b0f5136970eb2f7c/src/rviz/mesh_loader.cpp#L660). For example, [in the mesh marker test](https://github.com/ros-visualization/rviz/blob/1e1e96d0f28f115b90136a79386f0e6f48d23bae/src/test/mesh_marker_test.cpp#L69), multiple markers are published with the same resource but the mesh is only loaded once. 
Robot links [also use `loadMeshFromResource`](https://github.com/ros-visualization/rviz/blob/4c8793b634f810721e5c590f260169940b7f301f/src/rviz/robot/robot_link.cpp#L598).

To understand what is different in your situation could you give some additional context?
What visualisation type is the MotionPlanning plugin? Can you create a reproducible example using one of the default plugins? Is there a chance that it's the moveit plugin itself that's not re-using meshes, or have you ruled that out already?

rhaschke, 2018-04-18:
The MotionPlanning plugin instantiates multiple `rviz::Robots` to yield a trail of the robot trajectory. Maybe it's not the meshes but the numerous geometry primitives that create their own meshes over and over...
What about all the other members in rviz::Robot, e.g. selection? Do the consume lots of memory?

simonschmeisser, 2018-05-11:
Do you have attached objects? Those are unfortunately not cached. Also robot markers do some strange thinks when you have more than one mesh in a link. But I'm sure there are further places that could be optimized ...

rhaschke, 2018-05-11:
I don't have attached objects. I think, there should be a lightweight `RobotTrail` class in `rviz`, derived from `rviz::Robot` to explicitly share all data between different robot visualizations along the trail.

simonschmeisser, 2018-05-14:
I can confirm this, adding a moveit_ros_visualization/Trajectory and enabling "Show Trail" will increase memory usage by about 40Mb for a "normal sized" robot. It is a MoveIt! based plugin though ... maybe we should track it there?

rhaschke, 2018-05-14:
As I said, the MoveIt plugin simply allocates `rviz::Robot` instances for each trail point.
Manually adding `RobotModel` displays I also see an increase of a few Mb per instance. 
Curiously, this memory is *not* freed when removing the `RobotModel` instances! This indicates a leak in `RobotModel`.
In MoveIt we continously allocate and free Robot instances...

a-price, 2018-08-11:
Not sure if this should go here or in the MoveIt plugin, but there appears to be a significant memory leak when using the `moveit_ros_visualization/Trajectory` plugin. Each of the memory stairsteps below corresponds to sending a new `DisplayTrajectory` to RViz

[image]

[image]

The trajectories are for a couple of Kuka IIWA-7's with 150 steps per trajectory.

rhaschke, 2018-08-22:
@wjwwood @dhood How can we progress on this issue?

rhaschke, 2018-10-02:
I tracked this issue down to the Material handling in `rviz::Robot`. Although, the material is stored as a SharedPtr, it's not actually freed, because another pointer instance is held by the `MaterialManager`.
If I will find some time next week, I will propose a PR to fix the issue.

PR #1294 (2019-02-17): fix material handling
This attempts to resolve #1222. As noted in https://github.com/ros-visualization/rviz/issues/1222#issuecomment-426464344, I was able to track down the "memory leak" to material handling of robot links. Each time, a new `rviz::RobotLink` was created, the corresponding link material(s) were registered again with the `Ogre::MaterialManager` and never released until rviz finished.

This PR addresses the issue by essentially rewriting material handling of robot links: Instead of registering materials with the `MaterialManager`, they are directly created using `new Ogre::Materi [... cut]
closed 2019-02-17 by rhaschke