ros-visualization/rviz #1226: The rviz is crashed with the error Segmentation fault (core dumped)

insistyxj, 2018-04-25:
insisyxj@insisyxj:~$ rosrun rviz rviz -l
[ INFO] [1524643034.028844916]: rviz version 1.12.15
[ INFO] [1524643034.028931147]: compiled against Qt version 5.5.1
[ INFO] [1524643034.028941748]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1524643034.113287263]: Creating resource group General
[ INFO] [1524643034.113554205]: Creating resource group Internal
[ INFO] [1524643034.113612817]: Creating resource group Autodetect
[ INFO] [1524643034.113669098]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[ INFO] [1524643034.113852078]: Registering ResourceManager for type Material
[ INFO] [1524643034.113968429]: Registering ResourceManager for type Mesh
[ INFO] [1524643034.114006904]: Registering ResourceManager for type Skeleton
[ INFO] [1524643034.114048299]: MovableObjectFactory for type 'ParticleSystem' registered.
[ INFO] [1524643034.114117164]: ArchiveFactory for archive type FileSystem registered.
[ INFO] [1524643034.114150481]: ArchiveFactory for archive type Zip registered.
[ INFO] [1524643034.114192806]: ArchiveFactory for archive type EmbeddedZip registered.
[ INFO] [1524643034.114220574]: DDS codec registering
[ INFO] [1524643034.114284198]: FreeImage version: 3.17.0
[ INFO] [1524643034.114305618]: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
[ INFO] [1524643034.114429823]: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
[ INFO] [1524643034.114526216]: PVRTC codec registering
[ INFO] [1524643034.114545060]: ETC codec registering
[ INFO] [1524643034.114574261]: Registering ResourceManager for type HighLevelGpuProgram
[ INFO] [1524643034.114594414]: Registering ResourceManager for type Compositor
[ INFO] [1524643034.114735538]: MovableObjectFactory for type 'Entity' registered.
[ INFO] [1524643034.114772760]: MovableObjectFactory for type 'Light' registered.
[ INFO] [1524643034.114799734]: MovableObjectFactory for type 'BillboardSet' registered.
[ INFO] [1524643034.114815538]: MovableObjectFactory for type 'ManualObject' registered.
[ INFO] [1524643034.114829237]: MovableObjectFactory for type 'BillboardChain' registered.
[ INFO] [1524643034.114841709]: MovableObjectFactory for type 'RibbonTrail' registered.
[ INFO] [1524643034.114943281]: *-*-* OGRE Initialising
[ INFO] [1524643034.114965445]: *-*-* Version 1.9.0 (Ghadamon)
[ INFO] [1524643034.115008362]: OverlayElementFactory for type Panel registered.
[ INFO] [1524643034.115041429]: OverlayElementFactory for type BorderPanel registered.
[ INFO] [1524643034.115074709]: OverlayElementFactory for type TextArea registered.
[ INFO] [1524643034.115117599]: Registering ResourceManager for type Font
[ INFO] [1524643034.115187607]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL
[ INFO] [1524643034.117728037]: Installing plugin: GL RenderSystem
[ INFO] [1524643034.117770993]: OpenGL Rendering Subsystem created.
[ INFO] [1524643034.138106456]: Plugin successfully installed
[ INFO] [1524643034.138175720]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_OctreeSceneManager
[ INFO] [1524643034.139028523]: Installing plugin: Octree Scene Manager
[ INFO] [1524643034.139063791]: Plugin successfully installed
[ INFO] [1524643034.139078278]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_ParticleFX
[ INFO] [1524643034.140126507]: Installing plugin: ParticleFX
[ INFO] [1524643034.140154687]: Particle Emitter Type 'Point' registered
[ INFO] [1524643034.140168199]: Particle Emitter Type 'Box' registered
[ INFO] [1524643034.140181104]: Particle Emitter Type 'Ellipsoid' registered
[ INFO] [1524643034.140193150]: Particle Emitter Type 'Cylinder' registered
[ INFO] [1524643034.140203727]: Particle Emitter Type 'Ring' registered
[ INFO] [1524643034.140215655]: Particle Emitter Type 'HollowEllipsoid' registered
[ INFO] [1524643034.140226799]: Particle Affector Type 'LinearForce' registered
[ INFO] [1524643034.140240043]: Particle Affector Type 'ColourFader' registered
[ INFO] [1524643034.140251529]: Particle Affector Type 'ColourFader2' registered
[ INFO] [1524643034.140263111]: Particle Affector Type 'ColourImage' registered
[ INFO] [1524643034.140274830]: Particle Affector Type 'ColourInterpolator' registered
[ INFO] [1524643034.140287426]: Particle Affector Type 'Scaler' registered
[ INFO] [1524643034.140299360]: Particle Affector Type 'Rotator' registered
[ INFO] [1524643034.140312048]: Particle Affector Type 'DirectionRandomiser' registered
[ INFO] [1524643034.140323410]: Particle Affector Type 'DeflectorPlane' registered
[ INFO] [1524643034.140334802]: Plugin successfully installed
[ INFO] [1524643034.140355972]: CPU Identifier & Features
[ INFO] [1524643034.140366887]: -------------------------
[ INFO] [1524643034.140377395]:  *   CPU ID: GenuineIntel: Intel(R) Core(TM) i5-7400 CPU @ 3.00GHz
[ INFO] [1524643034.140388450]:  *      SSE: yes
[ INFO] [1524643034.140398298]:  *     SSE2: yes
[ INFO] [1524643034.140408315]:  *     SSE3: yes
[ INFO] [1524643034.140417552]:  *      MMX: yes
[ INFO] [1524643034.140427296]:  *   MMXEXT: yes
[ INFO] [1524643034.140437287]:  *    3DNOW: no
[ INFO] [1524643034.140446703]:  * 3DNOWEXT: no
[... 165 lines cut ...]
[ INFO] [1524643034.155236577]: Finished parsing scripts for resource group Internal
[ INFO] [1524643034.155248898]: Creating resources for group Internal
[ INFO] [1524643034.155258986]: All done
[ INFO] [1524643034.155268771]: Parsing scripts for resource group rviz
[ INFO] [1524643034.155793970]: Parsing script glsl120.program
[ INFO] [1524643034.156722998]: Parsing script nogp.program
[ INFO] [1524643034.157267867]: Parsing script point_cloud_tile.material
[ INFO] [1524643034.157446539]: Parsing script default_pick_and_depth.material
[ INFO] [1524643034.157631679]: Parsing script point_cloud_point.material
[ INFO] [1524643034.157787328]: Parsing script indexed_8bit_image.material
[ INFO] [1524643034.157873989]: Parsing script point_cloud_box.material
[ INFO] [1524643034.158024031]: Parsing script point_cloud_flat_square.material
[ INFO] [1524643034.158228511]: Parsing script point_cloud_square.material
[ INFO] [1524643034.158395545]: Parsing script point_cloud_sphere.material
[ INFO] [1524643034.158567832]: Parsing script liberation_sans.fontdef
[ INFO] [1524643034.158643625]: Finished parsing scripts for resource group rviz
[ INFO] [1524643034.158658847]: Creating resources for group rviz
[ INFO] [1524643034.158684066]: All done
[ INFO] [1524643034.159727065]: GLRenderSystem::_createRenderWindow "OgreWindow(1)", 100x30 windowed  miscParams: FSAA=4 contentScalingFactor=1.000000 externalGLControl= externalWindowHandle=83886099 parentWindowHandle=83886099 
[ INFO] [1524643034.163021212]: GLXWindow::create used FBConfigID = 112
[ INFO] [1524643034.202073676]: Mesh: Loading rviz_sphere.mesh.
[ INFO] [1524643034.206100240]: Texture: SelectionRect0Texture: Loading 1 faces(PF_R8G8B8A8,1x1x1) Internal format is PF_A8R8G8B8,1x1x1.
[ INFO] [1524643034.207693552]: Mesh: Loading rviz_cylinder.mesh.
[ INFO] [1524643034.207874865]: Mesh: Loading rviz_cone.mesh.
Segmentation fault (core dumped)

agustincosta, 2018-09-10:
In my case this was solved updating the Mesa 3D graphics library. I am running ROS kinetic in Ubuntu 16.04 on a VM. If that is your case you should also disable hadware 3D acceleration.

This is a link on how to update the Mesa 3D graphics on Ubuntu 16.04: http://ubuntuhandbook.org/index.php/2018/01/how-to-install-mesa-17-3-3-in-ubuntu-16-04-17-10/

I hope this helps because I lost days troubleshooting this as a beginner in ROS and Ubuntu myself.

dhood, 2018-09-17:
A few other users have reported that the suggestion by @agustincosta resolved the issue for them, so I will close this issue. Thank you for the tip @agustincosta !

closed 2018-09-17 by dhood

wathabs, 2020-07-26:
[ INFO] [1595755350.997010289]: rviz version 1.12.17
[ INFO] [1595755350.997062773]: compiled against Qt version 5.5.1
[ INFO] [1595755350.997075880]: compiled against OGRE version 1.9.0 (Ghadamon)
QXcbConnection: XCB error: 148 (Unknown), sequence: 171, resource id: 0, major code: 140 (Unknown), minor code: 20
Segmentation fault (core dumped)

i had tried all the way but the problem does not solved