ros-visualization/rviz #1228: Segmentation Fault trying RVIZ in ROS

dunfrey, 2018-04-27:
I'm trying to use on UDOO QUAD the Semi-direct Visual Odometry (SVO) and ROS algorithm: https://github.com/uzh-rpg/rpg_svo/wiki

The ROS was successfully installed and the SVO as well.

When I need to run RVIZ, I do with the command $ rosrun rviz rviz -d <PATH TO rpg_svo>, as shown in this link: https://github.com/uzh-rpg/rpg_svo/wiki/Run-SVO-with-ROS

But I could not execute and fail: Segmentation Fault

Run roscore is ok
Execute SVO is ok
Run the RVIZ not ok

Does anyone have any ideas why this is?

wjwwood, 2018-04-27:
Please, include a backtrace of the segfault. We cannot do anything without more information.

Also, make sure something like `glxgears` works before trying to debug rviz. If you're video card is not sufficient or setup properly rviz will not work anyways.

rhaschke, 2021-02-06:
Closing due to inactivity. Please feel free to reopen when you have more infos to reproduce.

closed 2021-02-06 by rhaschke