ros-visualization/rviz #1240: Could not find a package configuration file provided by "angles"

wmx-github, 2018-05-11:
cmake output

/opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83: error: Could not find a package configuration file provided by "angles" with any of the following names: anglesConfig.cmake angles-config.cmake Add the installation prefix of "angles" to CMAKE_PREFIX_PATH or set "angles_DIR" to a directory containing one of the above files.  If "angles" provides a separate development package or SDK, be sure it has been installed. CMakeLists.txt:116 (find_package)

CMakeLists.txt 

find_package(catkin REQUIRED
  COMPONENTS
  angles
  cmake_modules
  geometry_msgs
  image_transport
  interactive_markers
  laser_geometry
  map_msgs
  message_filters
  nav_msgs
  pluginlib
  python_qt_binding
  resource_retriever
  rosbag
  rosconsole
  roscpp
  roslib
  rospy
  sensor_msgs
  std_msgs
  std_srvs
  tf
  urdf
  visualization_msgs
  urdfdom_headers
)

how can i solve this problem ? 843230304@qq.com  or  wmx-github

wjwwood, 2018-05-11:
Please take more time to properly format your questions and add more details, like how you produced this error.

Also please ask questions on answers.ros.org, see: https://wiki.ros.org/Support

----

You are missing dependencies. Please refer to the ROS wiki or existing questions of answers.ros.org to see how to build packages and get their dependencies, e.g.: https://answers.ros.org/question/252478/how-to-build-a-package-from-source-on-ubuntu-mate-1604-lts-and-ros-kinetic/?answer=252502#post-id-252502

closed 2018-05-11 by wjwwood

avgprog, 2022-01-01:
I had a similar error. I think we need to install this [package](https://github.com/ros/angles).

felixf4xu, 2023-07-13:
@avgprog you are right. https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml#L301