ros-visualization/rviz #1247: include image_view.h

OAkyildiz, 2018-05-31:
Is 'image_view.h' exported as a part of librviz or anything else? I am trying to include it in the project, but it is not in the 'include' folder of the installed package. I can just copy the header and the source, but that is not a very clean way.

wjwwood, 2018-05-31:
It's not public no. You probably should not use the interface it describes since it could change at any time.

We could consider making it public, but you'll need to describe your use case and why you need it a little more and then potentially help with a pull request to make it a public header (mostly just installing it and making sure it doesn't change the run dependencies).

OAkyildiz, 2018-06-01:
Will do, as soon as I get a chance to. In the meantime, I would say, I want similar implementation and behavior to RenderPanel as a part of my GUI.

OAkyildiz, 2018-06-04:
Will do as soon as I get a chance to. In the mean

closed 2018-06-04 by OAkyildiz