ros-visualization/rviz #1249: rviz RobotModel broken in melodic: <visual><origin> not used

andreasBihlmaier, 2018-05-31:
The following two pictures will likely tell the whole story:
Kinetic on 16.04
 [image]
Melodic on 18.04
[image]

It appears that the <origin> tags within <visual> are not taken into account.

URDF:
```
<?xml version="1.0" ?>
<robot name="abc">
  <link name="a_link">
    <visual name="a_visual">
      <origin xyz="-0.5 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rviz_melodic_urdf_issue/a.dae" scale="1.0 1.0 1.0" />
      </geometry>
    </visual>
  </link>
  <joint name="a_b_joint" type="fixed">
    <parent link="a_link"/>
    <child link="b_link"/>
    <origin xyz="0 0 0" rpy="0 0 0" />
  </joint>
  <link name="b_link">
    <visual name="b_visual">
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rviz_melodic_urdf_issue/b.dae" scale="1.0 1.0 1.0" />
      </geometry>
    </visual>
  </link>
  <joint name="b_c_joint" type="fixed">
    <parent link="b_link"/>
    <child link="c_link"/>
    <origin xyz="0 0 0" rpy="0 0 0" />
  </joint>
  <link name="c_link">
    <visual name="c_visual">
      <origin xyz="0.5 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rviz_melodic_urdf_issue/c.dae" scale="1.0 1.0 1.0" />
      </geometry>
    </visual>
  </link>
</robot>
```

Clone https://github.com/andreasBihlmaier/rviz_melodic_urdf_issue into catkin workspace, build and run
```
roslaunch urdf_tutorial display.launch model:=`rospack find rviz_melodic_urdf_issue`/abc.urdf
```

wjwwood, 2018-05-31:
Thanks for the detailed report and examples. We'll have a look as soon as possible.

wjwwood, 2018-05-31:
Also, if anyone else can spend time to confirm this bug and/or suggest a solution in a pull request that would be great.

ProfFan, 2018-06-07:
Can not reproduce on my Arch with latest melodic compiled from source.

dani9, 2018-06-08:
I have the same problem with ubuntu 1804.

rahulbhadani, 2018-06-21:
I am a facing similar issue and it may require urgent fixing.

Fkallstrom, 2018-06-30:
I have the same problem ubuntu 18.04

dhood, 2018-07-02:
I have been trying to reproduce this issue (thank you again for the package with the examples), but I haven't been able to. I've installed rviz 1.13.1 from debians (qt 5.9.5, ogre 1.9.0) on two fresh installations of ubuntu bionic. One machine has an nvidia graphics card and the other intel.

We understand that this is affecting a number of users so we do want to get it resolved. Does anyone have info that could help us understand what is different on your machines to these two fresh installations? Are your machines also new installations, or did you upgrade from older ubuntu installations? Can you upload the console/log output?

In the meantime, for those that are blocked by this issue and desperate for a workaround, it looks like a clean re-install of ubuntu would do the trick.

rahulbhadani, 2018-07-02:
@dhood , I have a clean fresh installation of ROS Melodic on several machines in last two weeks on Ubuntu 18.04. Those installations are fresh installation.I found the same problem on all of machines I installed ROS Melodic (with Ubuntu 18.04 LTS). As a workaround I had to download kinetic-devel branch of rviz  and compile from the source.

ProfFan, 2018-07-02:
@rahulbhadani Can you try to install ROS from source and see what happens? This would determine if it is a problem in packaging.

dhood, 2018-07-02:
OK that's useful to know it has happened on clean installations.

Between kinetic-devel and melodic-devel branches there _has_ been [a change to the robot model display](https://github.com/ros-visualization/rviz/compare/kinetic-devel...melodic-devel#diff-66ee3f2e355c982cc6eb2beebff18c85R175) that might be what's causing the issue (though it's not clear to me yet why it wouldn't be showing on the machines I've tested with). If someone who has encountered/reproduced this issue could test if reverting https://github.com/ros-visualization/rviz/pull/1237 makes a difference, that would be helpful

rahulbhadani, 2018-07-02:
@dhood You can create a virtual machine and test it. That way you will know. @ProfFan Sure, I can do that, but it won't happen immediately from my side, as something else will keep me occupying this following week.

dhood, 2018-07-02:
I have tested in a virtual machine also but unfortunately cannot reproduce the original issue with a clean melodic installation, otherwise I would check the effect of reverting the PR myself.

We appreciate additional input/troubleshooting from you or anyone that gets a chance

rahulbhadani, 2018-07-02:
Okay, I am not sure how will I revert #1237 using command line.

heuristicus, 2018-07-05:
@rahulbhadani should be as simple as `git revert 40cc28`, since that pull request contained only a single commit.

dhood, 2018-07-09:
Appreciate the VM @andreasBihlmaier, that helped.

After a fair bit of investigation I tracked the cause down to differences in locale. On my machines, origins were being parsed as `origin: -0.500000, 0.000000, 0.000000` while on yours the equivalent would get parsed as `origin: -0,000000, 0,000000, 0,000000` (the commas are what tipped me off).

Turns out this is [a known issue](https://github.com/ros/urdfdom/issues/98) with `urdfdom`, which [has been addressed in `urdfdom_headers`](https://github.com/ros/urdfdom_headers/pull/42). It doesn't seem that the fix has been released into melodic, however. I will followup on that to try to get the fix into melodic if it's feasible (that's released as an upstream ubuntu package as opposed to a standard ROS package, so maybe not).

In the meantime a workaround is to unset/modify the `LC_NUMERIC` environment variable, e.g. `export LC_NUMERIC="en_US.UTF-8"`.

Can others please confirm that modifying the locale fixes it for them?

andreasBihlmaier, 2018-07-09:
@dhood Thank you very much for getting to the bottom of this nasty bug!

Setting locale as suggested fixes the issue for me.

Long ago that I last run into an `LC_*` related bug ... computer history's sins are haunting us.

rhaschke, 2019-02-17:
Closing this as it is fixed (but not yet released) in `urdfdom`.

closed 2019-02-17 by rhaschke