ros-visualization/rviz #1251: [WIP] rviz on Ogre 1.11

ProfFan, 2018-06-07:
Today I tried to run rviz on Arch Linux with ROS Melodic. It turns out that the current 1.13.1 release is not compatible with Ogre 1.11, which is the default for Arch Linux. Most notable problems are listed below for your information:

1. The ogre `.fontdef` shipped with Gazebo is non-conformant. Now the font name needs to be double-quoted, i.e., `font "Liberation Sans"`
2. Resources needs to be checked for existence before loading as a new resource. otherwise the engine will complain and die immediately.
```
[ERROR] [1528360037.402627853]: ItemIdentityException: Resource with the name package://franka_description/meshes/visual/link0.dae already exists. in ResourceManager::add at /home/fan/ARCHBUILD/ogre/src/ogre-1.11.0/OgreMain/src/OgreResourceManager.cpp (line 155)
[ERROR] [1528360037.403241901]: Could not load model 'package://franka_description/meshes/visual/link0.dae' for link 'panda_link0': ItemIdentityException: Resource with the name package://franka_description/meshes/visual/link0.dae already exists. in ResourceManager::add at /home/fan/ARCHBUILD/ogre/src/ogre-1.11.0/OgreMain/src/OgreResourceManager.cpp (line 155)
[ERROR] [1528360037.425471394]: ItemIdentityException: Resource with the name package://franka_description/meshes/visual/link0.dae already exists. in ResourceManager::add at /home/fan/ARCHBUILD/ogre/src/ogre-1.11.0/OgreMain/src/OgreResourceManager.cpp (line 155)
[ERROR] [1528360037.426065632]: Could not load model 'package://franka_description/meshes/visual/link0.dae' for link 'panda_link0': ItemIdentityException: Resource with the name package://franka_description/meshes/visual/link0.dae already exists. in ResourceManager::add at /home/fan/ARCHBUILD/ogre/src/ogre-1.11.0/OgreMain/src/OgreResourceManager.cpp (line 155)
```
3. By the same principle, resources need to be referenced by `SharedPtr`s, not raw pointers. The original way of taking the raw pointer out of `SharedPtr` causes double-free errors.
```
[ERROR] [1528361866.893538839]: RuntimeAssertionException: attempting to remove unknown resource: mesh_resource_marker_0package://franka_description/meshes/visual/hand.daeMaterial1 in group General in remove at /home/fan/ARCHBUILD/ogre/src/ogre-1.11.0/OgreMain/src/OgreResourceManager.cpp (line 308)
terminate called after throwing an instance of 'Ogre::RuntimeAssertionException'
  what():  RuntimeAssertionException: attempting to remove unknown resource: mesh_resource_marker_0package://franka_description/meshes/visual/hand.daeMaterial1 in group General in remove at /home/fan/ARCHBUILD/ogre/src/ogre-1.11.0/OgreMain/src/OgreResourceManager.cpp (line 308)
```
Currently trying hard to patch it to work.

CatchMeFastFat, 2018-09-03:
@ProfFan  Hello,

Could you help with:

I have changed font "Liberation Sans" `/opt/ros/melodic/share/rviz/ogre_media/fonts/liberation_sans.fontdef` with ", but the error still the same:

    terminate called after throwing an instance of 'Ogre::Exception'
      what():  : Could not find font Liberation Sans in MovableText::setFontName

regards,

ProfFan, 2018-09-04:
@CatchMeFastFat You can try use my fork in the meantime.

rhaschke, 2019-08-31:
See #1242 and #1252.

closed 2019-08-31 by rhaschke