ros-visualization/rviz #1254: Problem compiling "ogre_helpers/point_cloud.cpp" with GCC 8.1.1 on ArchLinux

ziokuroi, 2018-06-11:
The error at "point_cloud.cpp:340"
`src/rviz/ogre_helpers/point_cloud.cpp:340:52: error: no matching function for call to ‘rviz::PointCloudRenderable::setMaterial(const String&)’
     (*it)->setMaterial(current_material_->getName());`
I googled, but found nothing about the problem.

ziokuroi, 2018-06-11:
: ) maybe this is a dead project, fu*k ROS & fu*k linux

closed 2018-06-11 by ziokuroi

Alessandro-Barbieri, 2020-02-23:
I have the same issue on gentoo

rhaschke, 2020-02-23:
Which Ogre Version do you use? If it's new than 1.9.1, you should try the noetic-devel branch, which is prepared for Ogre 1.10 .. 1.12.

Alessandro-Barbieri, 2020-02-23:
ogre-1.9.0
I opened a bug here
https://bugs.gentoo.org/710540

rhaschke, 2020-02-23:
The error reported on gentoo.org differs from the one reported here. There seems to be a type conflict because Ogre was compiled with double precision, while on Ubuntu it's compiled with single precision.
You should replace suspicious hard-coded `float` types with `Ogre::Real` in rviz. Please file a PR with all these required changes.