ros-visualization/rviz #1257: Esc shortcut to leave fullscreen prevents canceling tools from working

ipa-fez, 2018-06-11:
https://github.com/ros-visualization/rviz/blob/070835c426b8982e304b38eb4a9c6eb221155d5f/src/rviz/visualization_frame.cpp#L484

This QShortcut seems to prevent esc key-events from firing which means 

https://github.com/ros-visualization/rviz/blob/070835c426b8982e304b38eb4a9c6eb221155d5f/src/rviz/tool_manager.cpp#L133

never gets called, meaning that actions such as send goal or initial pose can no longer be aborted with esc.

rhaschke, 2019-08-31:
I guess, this was fixed via your #1256? If not, reopen this issue and provide a more detailed error description.

closed 2019-08-31 by rhaschke

ipa-fez, 2019-09-02:
This isn't fixed by #1256 .

Here's what's happening:
Old behaviour:
- Press g to start "send goal" tool
- click and hold left mouse button to start the send goal action
- Press ESC to cancel the action without sending a goal

This no longer works now since the ESC key was bound to "exit full screen" in addition to "cancel current action" and the "exit fullscreen" behaviour seems to prevent the key event from propagating to the tools.

A workaround to cancel started actions right now is to press "I" to switch to the interaction tool which will also cancel a started "send goal" action without sending the goal.

Possible solutions would be to:
- Remove the ESC shortcut to leave fullscreen (leaving F11 as a toggle)
- Making sure the ESC key propagates to the tools so when rviz is not fullscreened, actions are canceled and when rviz is fullscreened, either exit fullscreen or both exit fullscreen and cancel any active actions.

rhaschke, 2019-09-02:
Thanks for describing the issue in more detail. When do you think the _old behavior_ was working?
According to the commit log, the ESC shortcut to leave fullscreen was introduced together with fullscreen mode. So I guess, you refer to a version w/o the fullscreen feature?

ipa-fez, 2019-09-03:
Exactly, when the version with fullscreen support was released, the bug was introduced.