ros-visualization/rviz #1258: i cant get rviz working on my ubuntu consol on my windows pc

zachcc99, 2018-06-11:
this is the error i keep getting how do i fix

zach@DESKTOP-SOP45BA:~$ rosrun rviz rviz
QXcbConnection: Could not connect to display
Aborted (core dumped)

zachcc99, 2018-06-12:
i managed to stop geting the other error but this one is now coming up
root@DESKTOP-SOP45BA:~#  xvfb-run rosrun rviz rviz
[ INFO] [1528798747.702760500]: rviz version 1.12.16
[ INFO] [1528798747.703637900]: compiled against Qt version 5.5.1
[ INFO] [1528798747.704173700]: compiled against OGRE version 1.9.0 (Ghadamon)
Unsupported screen format: depth: 8, red_mask: 0, blue_mask: 0
Unsupported screen format: depth: 8, red_mask: 0, blue_mask: 0
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
Unsupported screen format: depth: 8, red_mask: 0, blue_mask: 0
QWidget::paintEngine: Should no longer be called
QPainter::begin: Paint device returned engine == 0, type: 1
QPainter::setPen: Painter not active
Segmentation fault (core dumped)

VictorLamoine, 2018-07-10:
RViz is not meant to run Windows though you might get it workings thanks to Xming, take a look here:http://wiki.ros.org/Installation/Windows

dhood, 2018-09-17:
closing because as mentioned Windows is not an officially supported platform

closed 2018-09-17 by dhood