ros-visualization/rviz #1260: Starting rviz in fullscreen

davidbsp, 2018-06-15:
Is there an option to start rviz in fullscreen (cmd line args or via saved configuration file) ?

zaid-g, 2018-06-27:
Looking for the same thing. Hope someone has an answer

VictorLamoine, 2018-07-10:
There is no way right now, full screen was added in this merge request: https://github.com/ros-visualization/rviz/pull/1017/files

There is a menu and a shortcut only at the moment

rhaschke, 2019-09-22:
Fixed via #1413: new --fullscreen cmdline option.

closed 2019-09-22 by rhaschke

edgariph98, 2021-11-21:
how would I be able to set the configuration to be full screen?

rhaschke, 2021-11-21:
> how would I be able to set the configuration to be full screen?

@edgariph98, as written in the previous post (and document with `--help`):
`rviz --fullscreen`

edgariph98, 2021-11-22:
> > how would I be able to set the configuration to be full screen?
> 
> @edgariph98, as written in the previous post (and document with `--help`): `rviz --fullscreen`

is there a way to set a flag from a config.rviz file to start the rviz in full-screen mode? I am currently loading the rviz visualization frame into a window and I only want to see the rendering window, is there another way to achieve this instead of full-screen mode?

rhaschke, 2021-11-22:
No, I don't think so.