ros-visualization/rviz #1262: High DPI scaling issues with Qt::AA_EnableHighDpiScaling (macbook pro retina ubuntu 16.04)

rlober, 2018-06-19:
**tl;dr** 
I am raising this issue as a follow up to #1077 which was partially fixed in #1078. `Qt::AA_EnableHighDpiScaling` causes `RenderWidget` resize event to be 1/2 the proper size. Proposed fix in ______.

**Setup:**
- macbook pro late 2011
- retina display 
- ubuntu 16.04
- branch: `kinetic-devel`
- rviz at commit: 87742e4b8f40d8fe7e2a6952065651266faf73ba
- rviz version 1.12.16
- compiled against Qt version 5.11.0
- compiled against OGRE version 1.9.0 (Ghadamon)

**Problem:**
Since Qt 5.6 ([see here](http://doc.qt.io/qt-5/highdpi.html#high-dpi-displays)), high DPI scaling can be handled automatically by setting the application attribute `Qt::AA_EnableHighDpiScaling` in the `main.cpp` of a Qt application.

``` c++
// Enable support for high DPI screens 
QGuiApplication::setAttribute(Qt::AA_EnableHighDpiScaling);
```

This makes gui design more homogeneous across screens (at least where pixel coordinates are concerned). 

Unfortunately when I enable this attribute and am using `librviz`, the `RenderWidget` only fills 1/4 of the `RenderPanel`:
[image]

The rest of the space is filled with whatever was behind the `RenderPanel` upon a resize event.

When I do not set the `AA_EnableHighDpiScaling` attribute the `RenderWidget` displays properly, but the rest of the gui (not shown) is far too small:
[image]

The image above is scaled here but on my screen it is 2x smaller than it should be. This is because `AA_EnableHighDpiScaling` is not set.  

**Resolution:**

I say this problem was partially fixed in #1078 because the `AA_EnableHighDpiScaling` is never used in `rviz` and the `RenderWidget` displays as expected. 

I traced the problem to the `resizeEvent()` handler in `RenderWidget`:
``` c++
void RenderWidget::resizeEvent(QResizeEvent *e)
{
  if( render_window_ )
  {
    // render_window_->writeContentsToFile() (used in
    // VisualizationFrame::onSaveImage()) does not work right for
    // window with an odd width, so here I just always force it to be
    // even.
    render_window_->resize( width() + (width() % 2), height() );
    render_window_->windowMovedOrResized();
  }
}
```
When using `AA_EnableHighDpiScaling`, the functions `width()` and `height()` return the widget width and height with respect to a virtual pixel coordinate system and not the real pixel coordinate system. The `Ogre::RenderWindow` on the other hand is expressed in the real pixel coordiante system (I think...) and therefore only renders at 1/`devicePixelRatio()`.

I have proposed a fix in ______ which simply multiplies the width and height by the pixel ratio:

``` c++
render_window_->resize( width()*pixel_ratio_ + (width()*pixel_ratio_ % 2), height()*pixel_ratio_ );
``` 
This results in a properly scaled `RenderWidget`:
[image]

When not using `AA_EnableHighDpiScaling`, the widget fills the panel but it the whole gui is half the size, which is the expected behavior on high DPI screens.

mgrrx, 2018-07-23:
I tried the proposed fix on my Mac and it seems to work. Not sure if it is related to the issue described here when I try to do e.g. a Pose Estimate it only works in the upper right corner of the rendering area.

tompe17, 2018-09-24:
I see this issue on Ubuntu 18.04 and melodic. Did not have the issue on 16.04 and kinetic.

ahoarau, 2018-11-29:
Same issue with a Dell XPS 4k (16.04 / kinetic). https://github.com/ros-visualization/rviz/pull/1263 fixes it.

rhaschke, 2019-08-27:
Fixed via #1263.

closed 2019-08-27 by rhaschke