ros-visualization/rviz #1265: launching rviz issue: Segmentation fault (core dumped)

lulienhsi, 2018-07-13:
While using the command `rosrun rviz rviz` , the problem below happened. 
```
[ INFO] [1531463941.960011259]: rviz version 1.13.1
[ INFO] [1531463941.960058803]: compiled against Qt version 5.9.5
[ INFO] [1531463941.960072735]: compiled against OGRE version 1.9.0 (Ghadamon)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1531463942.079236403]: Stereo is NOT SUPPORTED
[ INFO] [1531463942.080276301]: OpenGl version: 3 (GLSL 1.3).
libpng warning: iCCP: known incorrect sRGB profile
Segmentation fault (core dumped)
```
I am using ubuntu 18.04 + ROS Melodic, did anyone confronted this problem before?

dhood, 2018-07-13:
Could you try updating all of your packages?

bgheneti, 2018-07-16:
fixed it for me

wjwwood, 2018-07-17:
@lulienhsi I'm going to close this since there's a recommended workaround and someone else seems to think it helped. Please comment here if it's not fixed yet and I'll reopen it.

closed 2018-07-17 by wjwwood

lulienhsi, 2018-07-17:
yeah, I updated them and it fixed, thanks for all the help!

tanyouliang95, 2019-02-26:
Hi I also have the same issue with the same printout, when i clicked on some keys on my keyboard, seg fault will happen. I tried to update rviz related packages but still it doesnt work. Would like to seek some help on this, or which pkgs are updated specifically?

rhaschke, 2019-02-26:
Please try to update all Ubuntu packages.

ali-mubarak001, 2025-07-24:
Hey, I have a similar issue on my Mac + Docker (Ubuntu 20.04):
root@dc8e4213570a:/opt# rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [1753381512.576663637]: rviz version 1.14.26
[INFO] [1753381512.577108304]: compiled against Qt version 5.12.8
[INFO] [1753381512.577158512]: compiled against OGRE version 1.9.0 (Ghadamon)
[INFO] [1753381512.598599387]: Forcing OpenGl version 0.
Segmentation fault
root@dc8e4213570a:/opt#