ros-visualization/rviz #1267: Not showing correctly the orientation of the poses of a nav_msgs/Path when described by an intermediate ref frame

joselusl, 2018-07-13:
I am using ROS kinetic in Ubuntu 16.04 and the most updated version of rviz that comes from apt.
I have two reference frames: parent_a_link, and parent_b_link. I also have the tf transformation between parent_a_link and parent_b_link. This transformation includes a rotation and a translation.
I publish a nav_msgs/Path message as path with frame_id parent_b_link.
If my Global Options -> Fixed Frame is parent_b_link, everything works fine.
But if my Global Options -> Fixed Frame is parent_a_link, then the orientation of the poses of the path is not correctly displayed. It seems that the displayed orientation is the orientation of the poses in parent_b_link (and should be shown the orientation in parent_a_link).

v01d, 2018-09-12:
I'm seeing a similar problem with a point cloud, see my description here: #1085

rhaschke, 2019-09-24:
I cannot reproduce this issue with rviz 1.13.3: The path points and frames perfectly move with the TF transform:
[image]

Note that [`nav_msgs/Path`](http://docs.ros.org/melodic/api/nav_msgs/html/msg/Path.html) involves a global header and individual headers for all mentioned poses. However, only the global one is considered by rviz! I consider it a design flaw to have multiple headers in this message: it's not clear which one to use.