ros-visualization/rviz #1268: Visualization bug on MacOS

dawnos, 2018-07-14:
I'm trying to compile & run rviz (kinetic) on MacOS (High Sierra), just following [http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source](url). After some minor modification, I successfully compile it. But when running, it get some problem:
[image]

3/4 of the display area is red.

Does anyone meet the same problem? Any solution?

Thank you!

rlober, 2018-07-15:
Not sure about the red, but the screen size issue is related to this: #1262

dawnos, 2018-07-16:
Thanks @rlober  . #1262 looks similar. I've tried it as what you say, but it still not work (same as above):

[image]
[image]
[image]

rlober, 2018-07-18:
Can you try adding `QGuiApplication::setAttribute(Qt::AA_EnableHighDpiScaling);` [here](https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/main.cpp#L37) ?

meyerj, 2018-07-25:
Same problem here. I also tried adding the line proposed in https://github.com/ros-visualization/rviz/issues/1268#issuecomment-405866942, without success. But it's not a big issue because the red area disappears and the scale becomes correct as soon as you move the camera or resize the window.

I am currently testing with rviz version 1.12.15 and Ogre 1.9 built from https://github.com/osrf/homebrew-simulation/blob/3ced67c30622d583d8df031dc7b91c5d8f6e7829/ogre1.9.rb on macOS Sierra.

rhaschke, 2019-08-31:
Fixed in #1263.

closed 2019-08-31 by rhaschke