ros-visualization/rviz #1270: kinetic branch no longer builds

jpapon, 2018-07-16:
kinetic-devel no longer builds due to linker errors from switching to tinyxml2.

I fixed this by pulling in the CMakeLists changes from #1237

wjwwood, 2018-07-16:
It looks like it is building on Kinetic:

http://build.ros.org/job/Kdev__rviz__ubuntu_xenial_amd64/42/

Are you just missing `tinyxml` on your system?

jpapon, 2018-07-19:
I should have included the error message - but it was trying to link to tinyxml2, even though the main CMakeLists links to tinyxml.  It works fine if you switch the find_package and link_libraries to tinyxml2.

I might have just been missing tinyxml though, and it just works with both versions?
I'll just close this since it looks like CI doesn't have any trouble with it...

closed 2018-07-19 by jpapon