ros-visualization/rviz #1271: Read-only propriety children values are not saved to the config

mmewen, 2018-07-25:
I figured that a read-write propriety in the property tree that is set descendant of a read-only propriety won't be saved to the `.rviz` config. 

Since [the doc of `Property::setReadOnly( bool )`](https://docs.ros.org/kinetic/api/rviz/html/c++/classrviz_1_1Property.html#a82fedd8e532997a3dd1cac0532a6e566) says that it's not recursive, I would consider its descendants to be saved.

Is it the expected behavior or is it a bug?
Thanks!

(I'm using RViz 1.12.16 kinetic)

rhaschke, 2019-08-31:
Looks like the current behavior is as follows:
If `is_read_only` or `!save`, the value of the property itself _is_ saved, but _all its children_ are not saved.

This, indeed, doesn't match the expectations raised from the doc you cited. 
I think a more sensible semantics would be as follows:
- `is_read_only` doesn't save the property itself, because the value, obviously, is _computed_
- A new `save_children` or `save_recursively` flag that decides whether children should be saved or not, i.e. allowing to recursively disable saving for all children.
- The `save` flag should allow disabling saving of the property itself, although the property is writable.
  However, this will not affect children. This is useful for help properties, like `All Enabled` in `TF`.

@wjwwood, any opinions on that proposal?

rhaschke, 2019-09-01:
@mmewen, I implemented the first two items from the list in #1411. Please, have a look.

closed 2019-09-01 by rhaschke

mmewen, 2019-09-02:
The semantics you've implement look good, thanks @rhaschke !