ros-visualization/rviz #1276: Constant CPU load

machinekoder, 2018-08-08:
I've noticed that the RViz preview uses ~25% of one core (Intel(R) Core(TM) i7-5600U CPU @ 2.60GHz) constantly, on an Intel(R) Pentium(R) CPU N4200 @ 1.10GHz this adds up to 75% CPU load of one core all the time.

This seems to be rather high to me for an "idle" RViz instance. Is this a known issue? If so, what is the best way to improve the performance of RViz?

PS: I've noticed that disable TF preview decreases the load by 3-4% percent and disabling the motion planning plugin brings it down to ~15%.

wjwwood, 2018-08-08:
It's not idle if you're displaying things like `tf`. It has to recieve and process messages constantly for most display types (joint state, tf, laser scan, point cloud, some kinds of markers, etc.). Even with nothing to do, it has a render loop fixed at ~33Hz, so that will take some effort, though it should not take as much as you've suggested.

I'm going to close this since it's a question right now. If you find something particular that's making it slow, I can reopen it.

closed 2018-08-08 by wjwwood

machinekoder, 2018-08-09:
I see. Are there any recommended best practices for optimizing the RViz performance?