ros-visualization/rviz #1281: Suggestion: Link to rviz_visual_tools in Marker documentation

felixvd, 2018-08-20:
I suggest adding a reference to [rviz_visual_tools](https://github.com/PickNikRobotics/rviz_visual_tools) on the http://wiki.ros.org/rviz/DisplayTypes/Marker page. It's not strictly the right place for it, but the page comes up on Google when looking for ROS visualization, and I believe it would be helpful for newcomers. I remember that I spent quite some time setting up my own convenience functions for publishing common markers, and that page was my main reference.

wjwwood, 2018-08-20:
Anyone can edit the wiki, if you found it useful, then referencing it is fine. Feel free to make that change yourself.

As you said, that might not be the best wiki page, but you guys will need to figure that out yourselves. Perhaps the https://wiki.ros.org/rviz/Tutorials/Markers%3A%20Basic%20Shapes tutorial instead. Or just the rviz wiki page.

closed 2018-08-20 by wjwwood

felixvd, 2018-08-28:
Done, thanks :)