ros-visualization/rviz #1283: Inverse Kinematics on rviz

tony23545, 2018-08-23:
Hi,

I am working on Fetch robot and motion planning. I have run into such a problem:

Given a serial of continue end-effector pose in work space, the serial of configuration in joint space are not continue (sometime flip).

I have noticed that when you drag the interactive marker on the end-effector in rviz, the robot's joints can always move in a continue trajectory. Does anyone have any idea what kind of IK algorithms rviz is using?

Thanks a lot!

mikeferguson, 2018-08-26:
@tony23545 generally it is poor form to post a question in multiple places. This is a duplicate of https://answers.ros.org/question/301384/inverse-kinematics-on-rviz/?comment=301600#post-id-301600 -- this type of question is definitely the type that should be asked on answers.ros.org

tony23545, 2018-09-05:
I think I have get the answer, thank you very much for your help and sorry for the duplicate question~

closed 2018-09-05 by tony23545