ros-visualization/rviz #1286: Nothing display after only change the view pitch.

Zander-yang, 2018-08-27:
I have drawn some strip lines in the rviz. When I move around with the mouse, sometimes the display area is all black. Is there any bugs in rviz? Or problems with my environment? I use a vm in my MacBook. The ubuntu version is 16.04. I will appreciate for any answers. Thanks a lot.
Bellowing screen prints show the problem. I only change the view pitch with keyboard input.
[image]
[image]
[image]

wjwwood, 2018-08-27:
I believe this is a duplicate of https://github.com/ros-visualization/rviz/issues/987 and in the past has been related to having very large transforms (using UTM zero as the world origin, for example). Please close this and add information to the other issue if you agree that it's a duplicate.

Zander-yang, 2018-08-28:
@wjwwood Thank you so much. It is actually a duplicate of #987 . I changed the frame to make the coordinates value small. It worked!

closed 2018-08-28 by Zander-yang