ros-visualization/rviz #1287: Lines display strangely

Zander-yang, 2018-08-30:
After I publish strip lines to rviz, the lines sometimes are blurry until the cursor slide over them. Is that a bug of rendering? There is a video in the zip. It's obviously at the beginning and end.
Looking forward to any replies. Thanks a lot.
[slide over.zip](https://github.com/ros-visualization/rviz/files/2334495/slide.over.zip)

rhaschke, 2019-08-31:
Could you provide a bag file and rviz config to easily reproduce this issue?
I never observed something like this.

mmurooka, 2020-08-19:
I got the sample problem.
Please unzip the attached file and run the following commands:
```
rosbag play rviz_issue_1287.bag -k
rviz -d rviz_issue_1287.rviz
```
[rviz_issue_1287.zip](https://github.com/ros-visualization/rviz/files/5095245/rviz_issue_1287.zip)

LINE_STRIP marker is published and displayed in Rviz. In the "Move Camera" mode in Rviz, the marker disappears when changing the viewpoint. In the "Interact" mode, the marker disappears but it reappear if the mouse cursor goes close to the marker.
[image]

My environment: Ubuntu16.04.1 / ROS Kinetic. Rviz is built from source and its version is [1.13.12](https://github.com/ros-visualization/rviz/commit/60f299878a50a40e65c5e8889e28f087f357f1fb). I also install ogre from source to fix [this issue](https://github.com/ros-visualization/rviz/issues/1082#issuecomment-534206651) and its version is [1.9.1](https://github.com/OGRECave/ogre/releases/tag/v1.9.1).

rhaschke, 2020-08-19:
Thanks for providing the bag file. I cannot reproduce this issue with a recent rviz (latest Melodic release).
Could you try to build the latest melodic-devel release and check with it?
Note, that you also need to build the melodic-devel branches of [TF1](https://github.com/ros/geometry) and [TF2](https://github.com/ros/geometry2) within your workspace to do so.

mmurooka, 2020-08-19:
I tried with the melodic-devel (the following catkin workspace) and the same problem still happens..
```
$ wstool info -t src
workspace: /home/mmurooka/ros/ws_rviz_melodic/src

 Localname S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 --------- - --- -------------- -----------  ---------------------------
 rviz        git melodic-devel  2b31dc4be241 github.com/ros-visualization/rviz.git
 geometry2   git melodic-devel  9b8022c30304 github.com/ros/geometry2.git
 geometry    git melodic-devel  0712a2cb808c github.com/ros/geometry.git
```
[image]

rhaschke, 2020-08-19:
Ok. Need to find a machine with older Xenial + Kinetic to test this tomorrow. But, I'm afraid the problem is related to your OpenGL lib. Could you please provide info on that (e.g. from rviz cmdline output).

mmurooka, 2020-08-20:
> provide info on that (e.g. from rviz cmdline output).

```
$ rviz
[ INFO] [1597888033.177879502]: rviz version 1.13.13
[ INFO] [1597888033.177934022]: compiled against Qt version 5.5.1
[ INFO] [1597888033.177951527]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1597888033.183421668]: Forcing OpenGl version 0.
[ INFO] [1597888033.901239841]: Stereo is NOT SUPPORTED
[ INFO] [1597888033.901307701]: OpenGl version: 4.6 (GLSL 4.6).
```

rhaschke, 2020-08-20:
I still cannot reproduce the issue, even on a Xenial/Kinetic setup. I guess, we have to blame a different version of OpenGL libs.
See also https://github.com/ros-visualization/rviz/issues/1508 and try `rviz --opengl 300`.

mmurooka, 2020-08-20:
I tried `--opengl 300` and `--opengl 210`, and the issue was reproduced in all cases..
[image]

rhaschke, 2020-08-20:
Sorry, then I don't have any more ideas.

martinakos, 2022-04-25:
Any progress on this? or workarounds? I have a simple visualization with LINE_LIST Markers and I can only see the lines in Rviz from some viewpoints. From other viewpoints the lines are not rendered, despite they should be visible. I'm using Noetic.

rhaschke, 2022-04-25:
I wasn't able to reproduce the effect so far. I you can in Noetic, please provide a yaml or rosbag with a corresponding marker message to reproduce the issue.

martinakos, 2022-04-26:
Another weird effect I've just realised, This is another drawing with multiple lines.  The lines don't render from certain view angles, but if I pass the mouse pointer over the unrendered lines while having the "interact" or "focus camera" tools the l lines are then redrawn and I can see them. This doesn't happen with "move camera" or "select" tools, ie. even if I move the cursor over the unrendered lines the line don't get drawn. This suggest that it may be an issue with the drawing event logic.

martinakos, 2022-04-29:
I see the LINE_LIST markers are drawn as flat tapes!! for a moment I thought that was because I was only setting scale.x and ignoring scale.y and scale.z. But after setting these nothing changed. In fact in the documentation says: "Line lists also have some special handling for scale: only scale.x is used and it controls the width of the line segments.". With width here I interpreted thickness of the line (as if the line was a cylindrical tube), but it's literally width of the tape, with thickness being apparently zero. I think the disappearing lines issue is because the visibility of the tape is fine when looking orthogonal to the width of the tape but when looking from the side the tape has zero thickness and it disappears.  

Could the LINE_LIST be reimplemented as a cylindrical tube rather than a flat (zero thickness) tape?

martinakos, 2022-05-06:
Maybe related #987

tigelbri, 2022-05-09:
I have the same issue. on some view angles the line is not rendered.

rhaschke, 2022-05-28:
I am able to reproduce the issue. All line strips essentially employ `Ogre::BillboardChain` to show a line that should always face the camera. However, this mechanism doesn't work for some reason. When moving the camera, the new camera pose isn't considered when drawing the BillboardChain.
In the `Interact` mode the camera pose is updated (and the line correctly displayed again) as soon as the mouse gets close to the line strip (into the bounding box?). This is related to the SelectionManager rendering the scene into a RenderTexture:
https://github.com/ros-visualization/rviz/blob/d8df909abe87899f66d1bf9471cfeb27395c4186/src/rviz/selection/selection_manager.cpp#L749

rhaschke, 2022-05-28:
The bug stems from OGRE 1.9. Updating to OGRE 1.12 resolves the issue.

rhaschke, 2022-06-04:
Trying to reproduce the issue once again, I first failed. Looks like the issue only occurs if the marker is not published regularly.
Or, phrased the other way around, a regular message updates the billboard line.

rhaschke, 2022-06-09:
This issue was fixed in OGRE 13.4.1: https://github.com/OGRECave/ogre/pull/2493
It won't fix for ROS1 anymore, as we are stuck with OGRE 1.9.0.

One can work around the issue by regularly updating the line strip marker.

jstmn, 2023-09-01:
experiencing this issue as well with ros noetic