ros-visualization/rviz #1290: Please help: rviz segmentation fault (core dumped)

chowjf, 2018-09-14:
This is how it looks like when I tried to run $rosrun rviz rviz -l (same happened to $rosrun rviz rviz)

[ INFO] [1536922162.289542339]: rviz version 1.12.16
[ INFO] [1536922162.289604496]: compiled against Qt version 5.5.1
[ INFO] [1536922162.289618713]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1536922162.405024650]: Creating resource group General
[ INFO] [1536922162.405124023]: Creating resource group Internal
[ INFO] [1536922162.405166631]: Creating resource group Autodetect
[ INFO] [1536922162.405357708]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[ INFO] [1536922162.405602989]: Registering ResourceManager for type Material
[ INFO] [1536922162.405642178]: Registering ResourceManager for type Mesh
[ INFO] [1536922162.405675746]: Registering ResourceManager for type Skeleton
[ INFO] [1536922162.405720885]: MovableObjectFactory for type 'ParticleSystem' registered.
[ INFO] [1536922162.405809718]: ArchiveFactory for archive type FileSystem registered.
[ INFO] [1536922162.405850964]: ArchiveFactory for archive type Zip registered.
[ INFO] [1536922162.405885707]: ArchiveFactory for archive type EmbeddedZip registered.
[ INFO] [1536922162.405917040]: DDS codec registering
[ INFO] [1536922162.405975999]: FreeImage version: 3.17.0
[ INFO] [1536922162.405999468]: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
[ INFO] [1536922162.406184228]: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
[ INFO] [1536922162.406221946]: PVRTC codec registering
[ INFO] [1536922162.406255122]: ETC codec registering
[ INFO] [1536922162.406286144]: Registering ResourceManager for type HighLevelGpuProgram
[ INFO] [1536922162.406331357]: Registering ResourceManager for type Compositor
[ INFO] [1536922162.406581600]: MovableObjectFactory for type 'Entity' registered.
[ INFO] [1536922162.406610091]: MovableObjectFactory for type 'Light' registered.
[ INFO] [1536922162.406625585]: MovableObjectFactory for type 'BillboardSet' registered.
[ INFO] [1536922162.406648432]: MovableObjectFactory for type 'ManualObject' registered.
[ INFO] [1536922162.406693288]: MovableObjectFactory for type 'BillboardChain' registered.
[ INFO] [1536922162.406726109]: MovableObjectFactory for type 'RibbonTrail' registered.
[ INFO] [1536922162.406846208]: *-*-* OGRE Initialising
[ INFO] [1536922162.406885923]: *-*-* Version 1.9.0 (Ghadamon)
[ INFO] [1536922162.406940327]: OverlayElementFactory for type Panel registered.
[ INFO] [1536922162.406973863]: OverlayElementFactory for type BorderPanel registered.
[ INFO] [1536922162.407005656]: OverlayElementFactory for type TextArea registered.
[ INFO] [1536922162.407051161]: Registering ResourceManager for type Font
[ INFO] [1536922162.407091097]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL
[ INFO] [1536922162.409887847]: Installing plugin: GL RenderSystem
[ INFO] [1536922162.409972869]: OpenGL Rendering Subsystem created.
[ INFO] [1536922162.429143875]: Plugin successfully installed
[ INFO] [1536922162.429252412]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_OctreeSceneManager
[ INFO] [1536922162.430871567]: Installing plugin: Octree Scene Manager
[ INFO] [1536922162.430941608]: Plugin successfully installed
[ INFO] [1536922162.430995943]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_ParticleFX
[ INFO] [1536922162.434041061]: Installing plugin: ParticleFX
[ INFO] [1536922162.434181453]: Particle Emitter Type 'Point' registered
[ INFO] [1536922162.434248633]: Particle Emitter Type 'Box' registered
[ INFO] [1536922162.434317149]: Particle Emitter Type 'Ellipsoid' registered
[ INFO] [1536922162.434372016]: Particle Emitter Type 'Cylinder' registered
[ INFO] [1536922162.434436615]: Particle Emitter Type 'Ring' registered
[ INFO] [1536922162.434493938]: Particle Emitter Type 'HollowEllipsoid' registered
[ INFO] [1536922162.434536972]: Particle Affector Type 'LinearForce' registered
[ INFO] [1536922162.434587727]: Particle Affector Type 'ColourFader' registered
[ INFO] [1536922162.434628666]: Particle Affector Type 'ColourFader2' registered
[ INFO] [1536922162.434668132]: Particle Affector Type 'ColourImage' registered
[ INFO] [1536922162.434716619]: Particle Affector Type 'ColourInterpolator' registered
[ INFO] [1536922162.434756299]: Particle Affector Type 'Scaler' registered
[ INFO] [1536922162.434803007]: Particle Affector Type 'Rotator' registered
[ INFO] [1536922162.434843783]: Particle Affector Type 'DirectionRandomiser' registered
[ INFO] [1536922162.434882077]: Particle Affector Type 'DeflectorPlane' registered
[ INFO] [1536922162.434918631]: Plugin successfully installed
[ INFO] [1536922162.434988896]: CPU Identifier & Features
[ INFO] [1536922162.435041740]: -------------------------
[ INFO] [1536922162.435093375]:  *   CPU ID: GenuineIntel: Intel(R) Core(TM) i5-4210U CPU @ 1.70GHz
[ INFO] [1536922162.435137948]:  *      SSE: yes
[ INFO] [1536922162.435187773]:  *     SSE2: yes
[ INFO] [1536922162.435229645]:  *     SSE3: yes
[ INFO] [1536922162.435269249]:  *      MMX: yes
[ INFO] [1536922162.435318598]:  *   MMXEXT: yes
[ INFO] [1536922162.435358783]:  *    3DNOW: no
[... 161 lines cut ...]
[ INFO] [1536922162.458127874]: Finished parsing scripts for resource group Internal
[ INFO] [1536922162.458154927]: Creating resources for group Internal
[ INFO] [1536922162.458178140]: All done
[ INFO] [1536922162.458201208]: Parsing scripts for resource group rviz
[ INFO] [1536922162.458929920]: Parsing script glsl120.program
[ INFO] [1536922162.460261454]: Parsing script nogp.program
[ INFO] [1536922162.461085772]: Parsing script point_cloud_tile.material
[ INFO] [1536922162.461367561]: Parsing script indexed_8bit_image.material
[ INFO] [1536922162.461512198]: Parsing script default_pick_and_depth.material
[ INFO] [1536922162.461792207]: Parsing script point_cloud_point.material
[ INFO] [1536922162.462030038]: Parsing script point_cloud_sphere.material
[ INFO] [1536922162.462236311]: Parsing script point_cloud_flat_square.material
[ INFO] [1536922162.462437721]: Parsing script point_cloud_square.material
[ INFO] [1536922162.462636125]: Parsing script point_cloud_box.material
[ INFO] [1536922162.462824353]: Parsing script liberation_sans.fontdef
[ INFO] [1536922162.462917765]: Finished parsing scripts for resource group rviz
[ INFO] [1536922162.462950855]: Creating resources for group rviz
[ INFO] [1536922162.462974869]: All done
[ INFO] [1536922162.464594390]: GLRenderSystem::_createRenderWindow "OgreWindow(1)", 100x30 windowed  miscParams: FSAA=4 contentScalingFactor=1.000000 externalGLControl= externalWindowHandle=58720275 parentWindowHandle=58720275 
[ INFO] [1536922162.467487667]: GLXWindow::create used FBConfigID = 251
[ INFO] [1536922162.522092027]: Mesh: Loading rviz_sphere.mesh.
[ INFO] [1536922162.528475376]: Texture: SelectionRect0Texture: Loading 1 faces(PF_R8G8B8A8,1x1x1) with 0 generated mipmaps from Image. Internal format is PF_A8R8G8B8,1x1x1.
[ INFO] [1536922162.531967024]: Mesh: Loading rviz_cylinder.mesh.
[ INFO] [1536922162.532274319]: Mesh: Loading rviz_cone.mesh.
Segmentation fault (core dumped)

dhood, 2018-09-14:
you could try the steps suggested by a user at https://github.com/ros-visualization/rviz/issues/1226#issuecomment-419907568

dhood, 2018-09-17:
thank you for reporting back @chowjf

wathabs, 2020-07-26:
[ INFO] [1595755350.997010289]: rviz version 1.12.17
[ INFO] [1595755350.997062773]: compiled against Qt version 5.5.1
[ INFO] [1595755350.997075880]: compiled against OGRE version 1.9.0 (Ghadamon)
QXcbConnection: XCB error: 148 (Unknown), sequence: 171, resource id: 0, major code: 140 (Unknown), minor code: 20
Segmentation fault (core dumped)

i had tried all the ways but the problem does not solved

hariharan382, 2020-08-18:
same problem,could not solve

qq0805, 2025-04-21:
你好，请问你解决了吗，我也遇到了QXcbConnection: XCB error: 147 (Unknown), sequence: 171, resource id: 0, major code: 141 (Unknown), minor code: 20
Segmentation fault (core dumped)
类似的错误