ros-visualization/rviz #1293: rviz crash at startup in in ros::WallTime::now()

elzKit, 2018-09-19:
I am on ubuntu 16_4 and ROS kinetic. 
glory@glory-ubuntu:~/catkin_ws/src/realsense$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Skylake Integrated Graphics (rev 06)
01:00.0 VGA compatible controller: NVIDIA Corporation GM107GLM [Quadro M1000M] (rev a2)

[ INFO] [1537377888.690014690]: rviz version 1.12.16
[ INFO] [1537377888.690049644]: compiled against Qt version 5.5.1
[ INFO] [1537377888.690058861]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1537377889.194281714]: Stereo is NOT SUPPORTED
[ INFO] [1537377889.194388398]: OpenGl version: 4.5 (GLSL 4.5).
Segmentation fault (core dumped)

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/opt/ros/kinetic/lib/rviz/rviz'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f925edc3f18 in ros::WallTime::now() () from /opt/ros/kinetic/lib/librostime.so
[Current thread is 1 (Thread 0x7f9262982840 (LWP 7952))]

jonbinney, 2018-09-21:
Are you on the lastest version of everything? Maybe try `sudo apt-get update && sudo apt-get dist-upgrade`?

Are you using any rviz plugins from source?

elzKit, 2018-09-21:
The sudo apt-get update && sudo apt-get dist-upgrade did solve the problem. Thanks!

closed 2018-09-21 by elzKit