ros-visualization/rviz #1298: RobotModel STL meshes displayed black with no lighting

VictorLamoine, 2018-10-10:
The meshes appear black and with no lighting in RViz:
[image]

# Test package
https://gitlab.com/InstitutMaupertuis/test_robot
```bash
roslaunch test_robot robot.launch
```
I discovered this when using the Fanuc packages (https://github.com/ros-industrial/fanuc/issues/246), you can also use them for testing.

# RViz
```bash
ii  ros-melodic-rviz                                     1.13.1-0bionic.20180907.011020                     amd64        3D visualization tool for ROS.
```
# Ogre
```bash
ii  libogre-1.9-dev                                      1.9.0+dfsg1-10                                     amd64        3D Object-Oriented Graphics Rendering Engine (development files)
ii  libogre-1.9.0v5:amd64                                1.9.0+dfsg1-10                                     amd64        3D Object-Oriented Graphics Rendering Engine (libraries)
```

# Graphic drivers
```bash
ii  libnvidia-cfg1-390:amd64                             390.77-0ubuntu0.18.04.1                            amd64        NVIDIA binary OpenGL/GLX configuration library
ii  libnvidia-common-390                                 390.77-0ubuntu0.18.04.1                            all          Shared files used by the NVIDIA libraries
ii  libnvidia-compute-390:amd64                          390.77-0ubuntu0.18.04.1                            amd64        NVIDIA libcompute package
ii  libnvidia-compute-390:i386                           390.77-0ubuntu0.18.04.1                            i386         NVIDIA libcompute package
ii  libnvidia-decode-390:amd64                           390.77-0ubuntu0.18.04.1                            amd64        NVIDIA Video Decoding runtime libraries
ii  libnvidia-decode-390:i386                            390.77-0ubuntu0.18.04.1                            i386         NVIDIA Video Decoding runtime libraries
ii  libnvidia-encode-390:amd64                           390.77-0ubuntu0.18.04.1                            amd64        NVENC Video Encoding runtime library
ii  libnvidia-encode-390:i386                            390.77-0ubuntu0.18.04.1                            i386         NVENC Video Encoding runtime library
ii  libnvidia-fbc1-390:amd64                             390.77-0ubuntu0.18.04.1                            amd64        NVIDIA OpenGL-based Framebuffer Capture runtime library
ii  libnvidia-fbc1-390:i386                              390.77-0ubuntu0.18.04.1                            i386         NVIDIA OpenGL-based Framebuffer Capture runtime library
ii  libnvidia-gl-390:amd64                               390.77-0ubuntu0.18.04.1                            amd64        NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD
ii  libnvidia-gl-390:i386                                390.77-0ubuntu0.18.04.1                            i386         NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD
ii  libnvidia-ifr1-390:amd64                             390.77-0ubuntu0.18.04.1                            amd64        NVIDIA OpenGL-based Inband Frame Readback runtime library
ii  libnvidia-ifr1-390:i386                              390.77-0ubuntu0.18.04.1                            i386         NVIDIA OpenGL-based Inband Frame Readback runtime library
ii  nvidia-compute-utils-390                             390.77-0ubuntu0.18.04.1                            amd64        NVIDIA compute utilities
ii  nvidia-dkms-390                                      390.77-0ubuntu0.18.04.1                            amd64        NVIDIA DKMS package
ii  nvidia-driver-390                                    390.77-0ubuntu0.18.04.1                            amd64        NVIDIA driver metapackage
ii  nvidia-kernel-common-390                             390.77-0ubuntu0.18.04.1                            amd64        Shared files used with the kernel module
ii  nvidia-kernel-source-390                             390.77-0ubuntu0.18.04.1                            amd64        NVIDIA kernel source package
ii  nvidia-opencl-dev:amd64                              9.1.85-3ubuntu1                                    amd64        NVIDIA OpenCL development files
ii  nvidia-prime                                         0.8.8.1                                            all          Tools to enable NVIDIA's Prime
ii  nvidia-settings                                      390.77-0ubuntu0.18.04.1                            amd64        Tool for configuring the NVIDIA graphics driver
ii  nvidia-utils-390                                     390.77-0ubuntu0.18.04.1                            amd64        NVIDIA driver support binaries
ii  xserver-xorg-video-nvidia-390                        390.77-0ubuntu0.18.04.1                            amd64        NVIDIA binary Xorg driver

```

VictorLamoine, 2018-10-23:
I understand a little bit more about how this happens.
Color values are not correctly parsed:
```bash
<material name="grey">
    <color rgba="0.3 0.3 0.3 1"/>
  </material>
```
Values smaller than 1.0 are just interpreted as zero!

So the only thing that is displayed correctly is combination of the primary RGB colors, any value with a value >=0.0  && <1.0 will be interpreted as zero.

[image]

Very probably related issue: https://github.com/ros-visualization/rviz/issues/1299

gavanderhoorn, 2018-10-23:
Could this be https://github.com/ros/urdfdom_headers/issues/45 somehow?

VictorLamoine, 2018-10-24:
Yes it is!
This bug will be fixed with the release of the latest `urdfdom_headers`.

Work-arround: https://github.com/ros-visualization/rviz/issues/1249#issuecomment-403351217

closed 2018-10-24 by VictorLamoine

gavanderhoorn, 2018-10-24:
I know :)

VictorLamoine, 2018-10-24:
Just summarizing for other people that might find this issue :wink:  Thanks for helping me

simonschmeisser, 2018-10-24:
Sorry to disappoint you guys, but looking here:
https://github.com/ros/urdfdom_headers/blob/e7e0972a4617b8a5df7a274ea3ba3b92e3895a35/urdf_model/include/urdf_model/color.h#L76
we can see that colors are not in fact fixed ...

nilsmelchert, 2018-10-24:
@VictorLamoine when will this fix be available in the apt tree?
Sorry, but I don't know what you mean by the latest `urdfdom_headers`.

simonschmeisser, 2018-11-14:
Small status update: https://github.com/ros/urdfdom_headers/issues/46 has been fixed by https://github.com/ros/urdfdom_headers/pull/47 and a test case using Dutch locale has been added in https://github.com/ros/urdfdom/pull/115

Now we just need this updated in the ubuntu packages

nilsmelchert, 2019-04-12:
Has it been updated in the ubuntu packages yet? Because I am still experiencing the same color issues with my robot as stated [here](https://github.com/ros-industrial/staubli_experimental/issues/30):

simonschmeisser, 2019-04-12:
If have filed a bug report asking for an update in Ubuntu Bionic but so far no reaction: https://bugs.launchpad.net/ubuntu/+source/urdfdom-headers/+bug/1817595

Please click: This bug affects me

Updated packages have been imported in Debian testing and made it into Ubuntu 19.04/Disco Dingo which is however not an LTS release!