ros-visualization/rviz #1301: PointCloud2 display is leaking memory when receiving messages with same timestamp

christian-rauch, 2018-10-12:
The `PointCloud2` display leaks memory when receiving consecutive `sensor_msgs::PointCloud2` with the same timestamp.
I assume this is related to the decay time. I admit that this is not a common scenario but it is still not what would be expected. The display should purge old point clouds when new messages with incremented sequence number arrive.

rhaschke, 2019-08-31:
Do you have time to provide a PR to fix this?

christian-rauch, 2019-08-31:
Not at the moment.
Can you confirm this issue?

rhaschke, 2019-09-01:
I only tried with PointCloud(1) display, which uses the same underlying code. But I actually couldn't reproduce the described problem. I observe the following memory evolution when varying decay time:

decay time | memory (MB)
----------------|------------------
0 | 120
10 | 229
1 | 208
0 | 203
10 | 230
0 | 218
10 | 230
0 | 220
10 | 230
0 | 227
10 | 230
0 | 226

Obviously, with a decay time of 10s I have constant upper limit of 230 MB.
If decay is reset to zero, I have varying values, which I don't quite understand yet.
I was sending 100 msgs/s with 4 points (modified [this example](https://github.com/ros-visualization/rviz/blob/d171d823965e456c6fec2286d6fa10541c84f1bb/src/test/send_two_clouds.py)).

christian-rauch, 2019-09-01:
The issue is actually not related to the timestamp of the message, but the ROS time published on `/clock`.
I can reproduce the issue with the `DepthCloud` display type when setting simulation time (`rosparam set /use_sim_time true`) and sending consecutive colour and depth messages with constant time in the header and on the clock topic. I assume that, because there is no time difference between consecutive messages, RViz will not remove the previous messages. The decay time and the queue size do not have an effect on this. RViz will gather ~30 MB per message leading to several GB within a minute.

christian-rauch, 2019-09-01:
I modified the same example to publish messages and clock with constant time:
```Python
#!/usr/bin/env python
import roslib
roslib.load_manifest('rviz')
import rospy
from sensor_msgs.msg import PointCloud
from geometry_msgs.msg import Point32
from rosgraph_msgs.msg import Clock
import numpy as np

rospy.init_node('send_two_clouds')
pub_cloud = rospy.Publisher("cloud", PointCloud, queue_size=1)
pub_clock = rospy.Publisher("/clock", Clock, queue_size=1)

tnow = rospy.Time.now()
cloud = PointCloud()
cloud.header.frame_id = "/base_link"
cloud.header.stamp = tnow

npoints = 640*240

for i in range(npoints):
    x,y,z = np.random.uniform(low=-1, high=+1, size=3)
    cloud.points.append(Point32(x, y, z))

while not rospy.is_shutdown():
    pub_cloud.publish(cloud)
    pub_clock.publish(clock=tnow)
```
This will lead to a memory leak in RViz with the PointCloud display when using simulation time (`rosparam set /use_sim_time true`).

rhaschke, 2019-09-01:
I would argue, that publishing a _constant_ time on `/clock` is evil. Just don't do that :wink:

christian-rauch, 2019-09-01:
Sure, this is definitely a corner case. You can look at it as some form of fuzzing :-)
If the decay time is set to 0 (only keep the oldest message), I would still expect that all other messages are discarded. I.e. the callback should clear the previously stored message if the new one is not older, or it should discard the new one because it is not newer. But it should not keep both in memory.

christian-rauch, 2019-09-01:
Fixed by #1412

closed 2019-09-01 by christian-rauch