ros-visualization/rviz #1302: libpng warnings at startup

VictorLamoine, 2018-10-16:
```bash
$ rosrun rviz rviz
[ INFO] [1539699041.413223554]: rviz version 1.13.1
[ INFO] [1539699041.413260869]: compiled against Qt version 5.9.5
[ INFO] [1539699041.413281630]: compiled against OGRE version 1.9.0 (Ghadamon)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1539699041.771737540]: Stereo is NOT SUPPORTED
[ INFO] [1539699041.772558821]: OpenGl version: 3 (GLSL 1.3).
libpng warning: iCCP: known incorrect sRGB profile
```

VictorLamoine, 2018-10-16:
It comes from ICC profiles inside PNG images. The PNG images have to be stripped from their profiles to avoid this kind of error.

```bash
find . -type f -name "*.png" -exec mogrify -strip {} {} \;
```

I still get the warnings after removing the profiles from the PNG images.

jarvisschultz, 2018-10-19:
@VictorLamoine, I do think that your description of what is wrong is correct, and I think that the command you've provided should fix the issue. When I used your command on my `rviz` PNG files, my warnings disappeared.

VictorLamoine, 2018-10-19:
You are right, I have tested again and it solves the issue, I have opened a pull request.
Thanks for testing

PR #1304 (2018-11-14): Remove ICC profiles in PNG images to silent libpng errors
Fixes #1302
closed 2018-11-14 by wjwwood