ros-visualization/rviz #1305: Ubuntu's rviz kinetic got SIGABRT in libOgreMain.so.1.9.0

theg4sh, 2018-10-23:
Got this fault:

```
rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp:630: virtual void Ogre::Node::setScale(const Ogre::Vector3&): Assertion `!inScale.isNaN() && "Invalid vector supplied as parameter"' failed.

Thread 1 "rviz" received signal SIGABRT, Aborted.
0x00007ffff69dc428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff69dc428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff69de02a in __GI_abort () at abort.c:89
#2  0x00007ffff69d4bd7 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x7ffff3c245e8 "!inScale.isNaN() && \"Invalid vector supplied as parameter\"", file=file@entry=0x7ffff3c24528 "/build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp", line=line@entry=630,
    function=function@entry=0x7ffff3c24a20 "virtual void Ogre::Node::setScale(const Ogre::Vector3&)") at assert.c:92
#3  0x00007ffff69d4c82 in __GI___assert_fail (assertion=0x7ffff3c245e8 "!inScale.isNaN() && \"Invalid vector supplied as parameter\"", file=0x7ffff3c24528 "/build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp", line=630,
    function=0x7ffff3c24a20 "virtual void Ogre::Node::setScale(const Ogre::Vector3&)") at assert.c:101
#4  0x00007ffff3a42d8f in Ogre::Node::setScale(Ogre::Vector3 const&) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#5  0x00007fffb69022f7 in rviz::ShapeMarker::onNewMessage(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void> > const> const&, boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void> > const> const&) () from /home/t4g/repos/cnord/catkin_ws/devel/lib/librviz_default_plugin.so
#6  0x00007fffb68fd1de in rviz::MarkerBase::setMessage(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void> > const> const&) () from /home/t4g/repos/cnord/catkin_ws/devel/lib/librviz_default_plugin.so
#7  0x00007fffb68d8eaa in rviz::MarkerDisplay::processAdd(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void> > const> const&) () from /home/t4g/repos/cnord/catkin_ws/devel/lib/librviz_default_plugin.so
#8  0x00007fffb68d85f8 in rviz::MarkerDisplay::processMessage(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void> > const> const&) () from /home/t4g/repos/cnord/catkin_ws/devel/lib/librviz_default_plugin.so
#9  0x00007fffb68d922c in rviz::MarkerDisplay::update(float, float) () from /home/t4g/repos/cnord/catkin_ws/devel/lib/librviz_default_plugin.so
#10 0x00007ffff7a808fd in rviz::DisplayGroup::update(float, float) () from /opt/ros/kinetic/lib/librviz.so
#11 0x00007ffff7b4c049 in rviz::VisualizationManager::onUpdate() () from /opt/ros/kinetic/lib/librviz.so
#12 0x00007ffff1f80d2a in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff1f8d5c8 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff1f81bb3 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff746405c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff7469516 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff1f5238b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff1fa75ed in QTimerInfoList::activateTimers() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff1fa7b29 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff14a9197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff14a93f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff14a949c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff1fa87cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff1f4fb4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007ffff1f57bec in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00000000004011cb in main ()
```

Run evironment:
```
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kinetic/lib/rviz/rviz...(no debugging symbols found)...done.
Starting program: /opt/ros/kinetic/lib/rviz/rviz --version
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdd772700 (LWP 13020)]
[ INFO] [1540317904.529952330]: rviz version 1.12.16
[ INFO] [1540317904.529993333]: compiled against Qt version 5.5.1
[ INFO] [1540317904.530004022]: compiled against OGRE version 1.9.0 (Ghadamon)
```

[rviz-sigabrt-inscale-isnan.coredump.gz](https://github.com/ros-visualization/rviz/files/2507381/rviz-sigabrt-inscale-isnan.coredump.gz)

wjwwood, 2018-10-23:
Thanks for the details summary, I don't have time to investigate, but perhaps it can be a lighting rod for other people with the issue and you guys can help one another.

theg4sh, 2018-10-24:
In additional, here is the listing of `ldd /opt/ros/kinetic/lib/rviz/rviz`:
https://gist.github.com/theg4sh/0e916104c3513896a354568560cb72a4

theg4sh, 2018-10-25:
Sorry for troubling, that was my bad. After ros update, I forgot to rebuild rviz from repository which i built earlier (a few months ago). Looks like works good now.

closed 2018-10-25 by theg4sh