ros-visualization/rviz #1306: Overscaled GUI on macOS Mojave

bgromov, 2018-10-24:
[image]

**Setup**:
 - MacBook Pro 15" 2018
 - macOS Mojave (10.14)
 - ROS Melodic built from source with [Mike Purvis' script](https://github.com/mikepurvis/ros-install-osx) with some modifications that are not related to graphics/GUI.
 - rviz 1.13.1 compiled against Qt 5.11.2 and OGRE 1.9.0

Rviz spits out a single warning to the terminal:

```
[ WARN] [1540412764.129368000]: GLHardwareVertexBuffer::lock - Error: failed to Discard the buffer. Try to recreate the buffer
```

The GUI itself, i.e. buttons etc., is overscaled but the 3D window is not. At the same time, the mouse works as if everything is of the proper scale, i.e. I can click on some controls, though sometimes the interface freezes completely.

I am also having this problem with Gazebo (9.4.1), but not with rqt.

[image]

[image]

What could be wrong?

Thanks

basaravia, 2018-11-04:
I have the same issue with my MacBook 2013

csobrinho, 2018-11-15:
Same issue with a MacBook Pro (13-inch, 2016) running Mojave [10.14.1 (18B75)]

zulufoxtrot, 2018-11-23:
Same issue here. Gazebo 8 and 9 from homebrew. MacOS Mojave. The interface looks exactly the same as in OP's screenshot. There is also an offset between the mouse pointer and the UI elements.

csvance, 2018-12-02:
Same issue here. If I drag rviz onto a non high-dpi monitor, all the widget sizes are corrected and the 3d window behaves mostly as expected after resizing.

arorasarthak, 2018-12-03:
Same issue here. However, it does work perfectly fine on non-retina resolutions with the mojave dark mode turned off.

csvance, 2018-12-03:
It's interesting that ViewsPanel and TimePanel, seems to scale ok, but the DisplaysPanel does not. They share a common loading mechanism (VisualizationFrame::loadPanels). There must be some difference between the panels that is causing the issue.

The toolbar (added with Qt method addToolBar) also has problems, but I am guessing that is unrelated because it doesn't share the same code path as the panels and mostly just uses Qt specific methods.

csvance, 2018-12-04:
With Qt 5.12 the interface scaling issues are fixed and dark mode works as expected:

[image]

However, still an issue with the 3d rendering part in the center.

bgromov, 2018-12-04:
@csvance Thanks, it worked. There are still problems with the palette, but I think it should be fixed somewhere in the RViz code. Btw, I have no issues with the 3D view at all (and I never had):

[image]

P.S. Building Qt from source takes eternity—it took two hours on i9. Hopefully the bottled version will appear soon

csvance, 2018-12-04:
I got it working as well!

[image]

Apparently its important that the terminal in which you launch rviz has the same DPI as where the window will open (which is based on the current rviz configuration)

So TLDR: if you want to run rviz on on your Retina display:

1. Open rviz, drag it to that display, and save your default configuration.
2. close rviz
2. Make sure your terminal you are launching rviz from is on the same display
3. Run rviz

If you want to run it on a non Retina display, do the same thing except with the non Retina display.

`bash-3.2$ lldb rviz
(lldb) target create "rviz"
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
Current executable set to 'rviz' (x86_64).
(lldb) break set -f render_widget.cpp -l 93
Breakpoint 1: where = librviz.dylib`rviz::RenderWidget::RenderWidget(rviz::RenderSystem*, QWidget*) + 298 at render_widget.cpp:93, address = 0x000000000014360a
(lldb) run
Process 74717 launched: '/opt/ros/melodic/bin/rviz' (x86_64)
[ INFO] [1543930649.869046000]: rviz version 1.13.1
[ INFO] [1543930649.869089000]: compiled against Qt version 5.12.0
[ INFO] [1543930649.869101000]: compiled against OGRE version 1.9.0 (Ghadamon)
2018-12-04 07:37:30.299136-0600 rviz[74717:1790254] libpng warning: iCCP: known incorrect sRGB profile
2018-12-04 07:37:30.300985-0600 rviz[74717:1790254] libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1543930650.748742000]: Stereo is NOT SUPPORTED
[ INFO] [1543930650.749654000]: OpenGl version: 2.1 (GLSL 1.2).
Process 74717 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
    frame #0: 0x000000010023660a librviz.dylib`rviz::RenderWidget::RenderWidget(this=0x000000010ea8dc00, render_system=0x000000010488bde0, parent=0x0000000104876500) at render_widget.cpp:93
   90  	  double pixel_ratio = window ? window->devicePixelRatio() : 1.0;
   91  	
   92  	#endif
-> 93  	  render_window_ = render_system_->makeRenderWindow(win_id, width(), height(), pixel_ratio);
   94  	}
   95  	
   96  	RenderWidget::~RenderWidget()
Target 0: (rviz) stopped.
(lldb) print pixel_ratio
(double) $0 = 2
(lldb) print width()
(int) $1 = 100
(lldb) print height()
(int) $2 = 30
(lldb) continue
Process 74717 resuming
2018-12-04 07:38:36.965758-0600 rviz[74717:1790254] libpng warning: iCCP: known incorrect sRGB profile
[ WARN] [1543930717.637057000]: GLHardwareVertexBuffer::lock - Error: failed to Discard the buffer. Try to recreate the buffer
Process 74717 exited with status = 0 (0x00000000) 
(lldb) run
Process 74722 launched: '/opt/ros/melodic/bin/rviz' (x86_64)
[ INFO] [1543930748.320632000]: rviz version 1.13.1
[ INFO] [1543930748.320682000]: compiled against Qt version 5.12.0
[ INFO] [1543930748.320694000]: compiled against OGRE version 1.9.0 (Ghadamon)
2018-12-04 07:39:08.613677-0600 rviz[74722:1796268] libpng warning: iCCP: known incorrect sRGB profile
2018-12-04 07:39:08.615598-0600 rviz[74722:1796268] libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1543930748.796662000]: Stereo is NOT SUPPORTED
[ INFO] [1543930748.797678000]: OpenGl version: 2.1 (GLSL 1.2).
Process 74722 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
    frame #0: 0x000000010023660a librviz.dylib`rviz::RenderWidget::RenderWidget(this=0x000000010cfd0580, render_system=0x000000010cfcdd00, parent=0x000000010cfcf980) at render_widget.cpp:93
   90  	  double pixel_ratio = window ? window->devicePixelRatio() : 1.0;
   91  	
   92  	#endif
-> 93  	  render_window_ = render_system_->makeRenderWindow(win_id, width(), height(), pixel_ratio);
   94  	}
   95  	
   96  	RenderWidget::~RenderWidget()
Target 0: (rviz) stopped.
(lldb) print width()
(int) $3 = 100
(lldb) print height()
(int) $4 = 30
(lldb) print pixel_ratio
(double) $5 = 1`

rhaschke, 2019-08-31:
Seems to be fixed.

closed 2019-08-31 by rhaschke