ros-visualization/rviz #1314: Drop Qt4 support for melodic

simonschmeisser, 2018-11-13:
Qt4 is no longer supported since December 2015 [1] and rviz has successfully worked with Qt5 for the duration of kinetic. I thinks it's time to remove all traces of Qt4 in melodic-devel and start supporting new features from Qt5. Especially compile-time checking of Signal-Slot connections comes to my mind here.

If nobody opposes I will open PRs for updated CMake and the code itself

[1] https://blog.qt.io/blog/2015/05/26/qt-4-8-7-released/

wjwwood, 2018-11-13:
That's fine by me, as Melodic (and all target platforms for it) require Qt5:

http://www.ros.org/reps/rep-0003.html#melodic-morenia-may-2018-may-2023

Supporting Qt4 too was just a convenience for people who wanted to build the melodic version on older Ubuntu's (not an officially supported case).

Also, I'll point out that any changes to do this should avoid API breaks and try to avoid ABI breaks for the melodic branch. So that might be an issue for targeting melodic devel.

nathangeorge1, 2018-11-16:
Not sure if this is the right place, but would anyone know of any good resources to connect RViz with Qt? @simonschmeisser

wjwwood, 2018-11-16:
@nathangeorge1  In general, please ask questions like this on answers.ros.org.

There's a tutorial for using rviz in a custom Qt gui: https://wiki.ros.org/rviz/Tutorials/Librviz%3A%20Incorporating%20RViz%20into%20a%20Custom%20GUI

nathangeorge1, 2018-11-16:
will do @wjwwood; thanks

rhaschke, 2019-03-05:
@simonschmeisser, you are welcome to open PRs for this.

simonschmeisser, 2020-03-30:
Thanks for going ahead in `noetic-devel` @rhaschke ! I was planning on doing this today actually but now I'll try to do some modernization based on it

closed 2020-03-30 by simonschmeisser