ros-visualization/rviz #1316: ImportError: No module named PyQt5.QtCore

TS-YinJie, 2018-11-28:


TS-YinJie, 2018-11-28:
I am learning the urdf /Tutorials/Using Xacro to Clean Up a URDF File ,when I run the launch,an error appears:
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher", line 6, in <module>
    from python_qt_binding.QtCore import pyqtSlot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
    from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 271, in <module>
    getattr(sys, 'SELECT_QT_BINDING_ORDER', None),
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 117, in _select_qt_binding
    raise ImportError("Could not find Qt binding (looked for: %s):\n%s" % (', '.join(["'%s'" % b for b in binding_order]), '\n'.join(error_msgs)))
ImportError: Could not find Qt binding (looked for: 'pyqt', 'pyside'):
  ImportError for 'pyqt': No module named PyQt5.QtCore
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 150, in _load_pyqt
    _named_import('PyQt5.%s' % module_name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 130, in _named_import
    module = builtins.__import__(name)
ImportError: No module named PyQt5.QtCore

  ImportError for 'pyside': No module named PySide2.QtCore
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 182, in _load_pyside
    _named_import('PySide2.%s' % module_name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 130, in _named_import
    module = builtins.__import__(name)
ImportError: No module named PySide2.QtCore

ubuntu 16.04
ros kinetic

rhaschke, 2019-03-29:
Could you please verify that you have `ros-kinetic-python-qt-binding` (and all its dependencies) correctly installed?

rhaschke, 2019-08-30:
Closing due to inactivity.

closed 2019-08-30 by rhaschke