ros-visualization/rviz #1324: Rviz slow rendering on MacOS Mojave

b-hahn, 2019-01-28:
I compiled ROS on MacOS Mojave 10.14.2 and can run Rviz (1.13.1, Qt 5.12, Ogre 1.9.0) but unfortunately rendering of even the empty default scene with a grid runs very slowly, at no more than around 1 frame per second. I have a 2015 Macbook with an Intel Iris 6100 so I believe that simple geometry should at least render somewhat fluidly. 
Is there some way to ensure that my GPU is being utilized? Or can I really expect no more than 1 FPS on my machine for even the simplest scenes?

Thanks a lot!

rhaschke, 2019-03-05:
Sorry, currently we have too few reviewers to handle your issue.

smnogar, 2019-03-22:
I have noticed this too. I have similar issues with Gazebo (and found a [related bug report](https://bitbucket.org/osrf/gazebo/issues/2579/system-wide-latency-induced-by-gazebo-960). I have tested it on multiple desktops with discrete graphics and performance is definitely degraded. This may be due to something introduced in macOS 10.14.2.

rhaschke, 2019-08-31:
Closing. Probably unrelated to rviz.

closed 2019-08-31 by rhaschke