ros-visualization/rviz #1329: volunteering for maintainer

rhaschke, 2019-02-08:
Dear @wjwwood,

thanks for pointing out https://discourse.ros.org/t/rviz-maintenance/6410. As stated there, I volunteer hereby to assist in maintaining this package. As you obviously turned to ROS2, there is some need to ensure progress on rviz for ROS1.
At ROSCon we discussed that it would be nice to separate the core of rviz and ROSx specific stuff. I won't have resources to work on this either. I can only promise to review and merge PRs, prepare releases, and thus help keeping rviz for ROS1 alive. I would be more than happy if other people follow and help to maintain this package in future (@simonschmeisser?).

rhaschke, 2019-02-10:
@wjwwood, you seem to develop rviz from osx as well. Is there any CI framework configured to test rviz PRs against osx?. Currently, you seem to use the ROS buildfarm only for CI testing. I think, Travis provides an osx environment as well. How do you configure/install ROS on your osx system?

wjwwood, 2019-02-11:
> Thanks for pointing out https://discourse.ros.org/t/rviz-maintenance/6410. As stated there, I volunteer hereby to assist in maintaining this package.

Great! Thanks :)

> At ROSCon we discussed that it would be nice to separate the core of rviz and ROSx specific stuff. I won't have resources to work on this either. I can only promise to review and merge PRs, prepare releases, and thus help keeping rviz for ROS1 alive.

Sounds reasonable and worthwhile.

> @wjwwood, you seem to develop rviz from osx as well.

Sometimes, but often I fall back to Linux for simplicity when testing.

> Is there any CI framework configured to test rviz PRs against osx?

No, not for ROS 1, we do test the ROS 2 version on macOS and Windows, however.

> [Travis provides an osx environment](https://docs.travis-ci.com/user/multi-os/) as well.

I've found that to be slow, but it might be worth investigating again. Stability of Homebrew and the ROS install instructions for macOS were too much for me to keep up with.

> How do you configure/install ROS on your osx system?

I follow the guide on @mikepurvis's github repository https://github.com/mikepurvis/ros-install-osx (more or less) and build from source on top of that.

----

I appreciate you stepping up to help, I'll get back to you here soon with more info. Thanks again!

rhaschke, 2019-02-12:
@wjwwood, I'm looking forward to hear from you. Would be great, if I could already use some spare time at the upcoming weekend to progress with rviz.

wjwwood, 2019-02-13:
I've sent you an invite to a GitHub team, which once you accept it you will have access to merge pr's and close issues, etc.

If you feel comfortable, please just jump in, but I'll try to allocate some additional resources in the next few weeks to double check your work if you want that. Just @ me to get a spot check.

We do have a "Kanban" board here: https://waffle.io/ros-visualization/rviz

A useful thing to do if you have spare cycles is to triage the items in inbox and either move them to backlog until someone has time to investigate (labeling them as bug or enhancement), or "in progress" if someone is working on them or you start working on it, or into "review" if it's a pull request that just needs review. We also have some useful labels for keeping track of the status of various pr's and issues, see: https://github.com/ros-visualization/rviz/wiki/Maintainer-Guide#triage-of-issues

When you think it's time for a release, ping me and we'll speak off-line to coordinate that.

That's all I have for now, thanks again for offering to help out.

closed 2019-02-13 by wjwwood