ros-visualization/rviz #1330: rviz crashes when loading a robot model

lakshmip001, 2019-02-11:
I tried loading robot_model in rviz and it crashes with the following error
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(4:ItemIdentityException): An object of type 'Entity' with name 'Robot Link4' already exists. in SceneManager::createMovableObject at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreSceneManager.cpp (line 6832)
Aborted (core dumped)

Kindly please help me why I get this error and how i can solve it.

rhaschke, 2019-03-05:
Please provide more details. Which rviz and Ogre version are you using?
Please provide a link to the robot model.

lakshmip001, 2019-03-06:
RVIZ crashes when I try loading trajopt trajectory visualization in RVIZ along with the robotmodel. RVIZ version
 rviz version 1.12.16
[ INFO] [1551835254.835176866]: compiled against Qt version 5.5.1
[ INFO] [1551835254.835186669]: compiled against OGRE version 1.9.0 (Ghadamon)

rhaschke, 2019-03-06:
Sorry, but this is not enough information to reproduce the error.

lakshmip001, 2019-03-06:
May be the below link can provide more information
https://github.com/ros-industrial-consortium/trajopt_ros/issues/56

rhaschke, 2019-08-30:
Is this still relevant? Can you try to provide a yaml or bag file of the message causing this issue?
If already solved, please close the issue.

rhaschke, 2019-09-24:
Closing due to inactivity. Feel free to reopen the issue (and provide more information in this case).

closed 2019-09-24 by rhaschke