ros-visualization/rviz #1349: Path visualization is not working in rviz

poonam1120, 2019-03-01:
Hi,

I have created controller node for my drone. The drone is moving fine in gazebo and I am trying to visualize the path driven by drone in rviz on [using nav_msgs/Path].
Initially path is correct but gradually there is huge diffrenece in rviz and gazebo.
For Example:- If in gazebo, drone completed straight line then in rviz only half line path is visualized.
I suspect that there is some issue with the time. I have also seen some open issues on nav_msgs/path topic. Apology if it is repeated question.

Please help me to get exact path traversed by drone on rviz.

Thansk in advance.

rhaschke, 2019-03-05:
Sorry, unfortunately, we do not have enough maintainers for rviz to handle your issue.
Are you sure, all messages are correctly published?
Did you try to manually publish a path message? Is this correctly displayed?

poonam1120, 2019-03-11:
Yes, I can echo on path topic and see the data coming properly.
Only when the path is displayed on the rviz, there is some delay.
As of now, I am just living with this delay. T

rhaschke, 2019-08-30:
If you are still facing this problem, please provide source code to reproduce this issue.
Otherwise, close the issue.

rhaschke, 2019-09-24:
Closing due to inactivity. Feel free to reopen the issue (and provide more information in this case).

closed 2019-09-24 by rhaschke